lvgl/lvgl #148: Add Notifications / Title Bar

BenArtes, 2018-03-06:
In my application, I'm trying to add a Notifications / Title bar to all screens containing Time + Date + Battery.  Currently, after creating the initial screen, I'm adding labels for these features and decreasing the size of subsequent objects so that they appear.

Is there potentially a better way to implement this?

kisvegabor, 2018-03-07:
So you have a title bar with a label e.g. `17:38     06-03-2018     78%`. Let's say it is 20 px high.

And you wold like to create the content below it. Let's say you have 320x240 screen so for the content 220 px remains. 

If I understood well then you can do the following:
1. Create 1 title bar container (only one not on every screen)
2. Create 1 content container on every screen.
3. Set their height and position (20 px and 220 px). 
4. When you load a new screen you can move the title bar there with `lv_obj_set_parent(title_bar, new_screen)`

BenArtes, 2018-03-07:
Awesome.  I'd assumed I should be using a container for the title bar but hadn't noticed / realized I could move it between screens with `lv_obj_set_parent`.

Thanks. :+1:

closed 2018-03-07 by BenArtes

kisvegabor, 2018-03-07:
Your welcome!