lvgl/lvgl #150: Can not ignore the tearing effect

Vonger, 2018-03-07:
If without background, it works pretty good. With background, you can find in this gif.
[image]

kisvegabor, 2018-03-07:
Hi,

You can improve the quality by increasing the size of VDB (LV_VDB_SIZE in lv_conf.h)
If you set it to `LV_HOR_RES * LV_VER_RES` the screen will be rendered in one piece. It's like double buffering.

Vonger, 2018-03-07:
@kisvegabor emm, not working.
more buffering just improve the speed, but tearing is not disappear.
Looks like the tearing position is fixed at a height(maybe at 240?) very weird.

Vonger, 2018-03-07:
@kisvegabor current my screen is 480x800, once I set it to 480x240, the tearing effect is not showing anymore. Any idea?

Vonger, 2018-03-07:
I can confirm it is tearing at height = 250px~300px
Looks like first frame only render to around 250px then next frame render the full screen.

[image]

kisvegabor, 2018-03-07:
I'm not sure but I suppose it comes from the PC simulator's monitor driver. When the monitor driver  (monitor.c) copies the pixels (in monitor_flush()) to the frame buffer then there is no good triggering to not refresh the screen while copying.  
However it sets the `sdl_refr_qry` flag it remains true after the first run and maybe SDL will update  the screen and clear the flag while you are copying the next frame.

You can try to set `sdl_refr_qry = false` in the beginning of `monitor_flush` to be sure it won't be refreshed during the flush.

Vonger, 2018-03-07:
@kisvegabor I do not think so, please check this video: https://youtu.be/_MouNVwW46k
It is on a framebuffer device.

sirius65535, 2018-03-08:
@Vonger Your video is blocked？

Vonger, 2018-03-08:
@sirius65535 Updated 👍

kisvegabor, 2018-03-08:
I see. It's sure that if you create a (LV_HOR_RES * LV_VER_RES) sized VDB then the whole screen will be refreshed in one piece. So `disp_flush` will get the full screen in the buffer.  I think it comes from the asynchronous update of the frame buffer. 
I found it: https://stackoverflow.com/questions/13907471/linux-framebuffer-graphics-and-vsync

WangZhihua1998, 2018-03-09:
@kisvegabor Why the size should be LV_HOR_RES *LV_VER_RES rather than the answer above times (color depth)/8?

Vonger, 2018-03-09:
@WangZhihua1998 unit is pixel.

j123b567, 2018-03-14:
Here is nice example of linux framebuffer animation with FBIO_WAITFORVSYNC https://github.com/rst-/raspberry-compote/blob/master/fb/fbtestXIV.c
It is for Raspberry Pi,  but it should work on any LinuxFB. You can also try fbtestXIVb.c in the same repo. 

It is using real double buffering and just switching buffers at the right time. You can try this example on your platform if your driver supports wayting for vsync and tearing free animation. 

Next step can be to incorporate this mechanism to lvgl driver for LinuxFB

kisvegabor, 2018-07-15:
As there was no activity in this topic for a few months I close this issue.

If you made some progress or have further questions don't hesitate to comment and reopen the issue.

closed 2018-07-15 by kisvegabor