lvgl/lvgl #151: how to

sirius65535, 2018-03-08:


kisvegabor, 2018-03-08:
Hi,

Thank you! :)

Let's assume that you have (or can convert) the image to a raw pixel map. In this case you need to to create a very simple file system with some custom functions to read data from the Flash. 
Here is a Tutorial with a POSIX file handling: https://github.com/littlevgl/lv_examples/blob/master/lv_tutorial/6_images/lv_tutorial_images.c

As you can see only the most basic functions are required.

avafinger, 2018-03-09:
I have a similar requirement, but i was thinking to display the photo in real-time and not converting it to a file and later read it. In an X11 environment, i grab the photo in YUV format and convert it to jpeg and then display it. Maybe you already have a container where i could pass a buffer in RGB565 or RGB888 format and this container would invalidate the area and display the buffer?

sirius65535, 2018-03-12:
@kisvegabor Thanks for your guidance，I was busy these days and I'll try it later :)

sirius65535, 2018-03-12:
@avafinger Thank you！ But I'm not familiar with LittlevGL now，I'll have a try.

kisvegabor, 2018-03-12:
@avafinger 
There is no object directly for this purpose now but you create  it easily from an image. Just create an image descriptor manually: 
```c
lv_color_t img_pixels[20][20];
uint32_t i, j;
for(i = 0; i < 20; i ++) {
    for(j = 0; j < 20; j ++) {
        img_pixels[i][j] = LV_COLOR_YELLOW;
    }
}

lv_img_t img_dsc;
img_dsc.header.alpha_byte = 0;
img_dsc.header.chroma_keyed = 0;
img_dsc.header.format = LV_IMG_FORMAT_INTERNAL_RAW;
img_dsc.header.w = 20;
img_dsc.header.h = 20;

img_dsc.pixel_map = img_pixels;

img_pixels[5][10] = LV_COLOR_RED;
img_pixels[6][10] = LV_COLOR_RED;
img_pixels[8][10] = LV_COLOR_RED;

lv_obj_t * img_obj = lv_img_create(lv_scr_act(), NULL);
lv_img_set_src(img_obj, &img_dsc);
lv_obj_set_pos(img_obj, 50, 50);
```

It results this:

[image]

This way you can store the photo in a variable and display it as an image object.
If you refresh the `img_pixels` array just call `lv_obj_invalidate(img_obj);`

WangZhihua1998, 2018-03-19:
@sirius65535老哥，我能请教一下你是如何移植到f407上的吗？我用的是stm32f407vet6+ili9341，用卖家提供的的工程文件可以驱动屏幕，但是仅能在armcc5下编译，在keil中改为armcc6就各种报错。(但是5编译不了lvgl)我尝试改为hal库却怎么都驱动不了屏幕，不知道哪里出了问题。如果老哥有现成的资源能否共享一下:blush:

Masi2017, 2018-04-20:
@kisvegabor  I have the same requirement, we need to add camera in our device and user should be able to take a photo and see the existing photo gallery. I didn't get that without image converter tools how can I display taken photo by camera on display?
@avafinger Hi, were you able to display the photos?

avafinger, 2018-04-20:
@Masi2017 
@kisvegabor 

I have not finished it yet due to time constraint but it is all my fault.  I think i get what you mean by "converter tools", i think you need to convert it to 32 bit (raw format). And if your camera delivers JPEG images then you need to link with some libjpeg or build your own converter. I haven't got that far yet, still need to port lvgl to two nice LCDs (5" and 4.3") i just received. But i am not working with STM32 but with some powerfull ARM boards. To display a gallery i would suggest to add to lvgl as a new feature something like a listbox with loadable dynamic images ( a callback to load the image for each item as it is displayed ).

kisvegabor, 2018-04-20:
As @avafinger suggested the easiest way is to check how a converted image builds up and mimic it with a decoded camera image. The color depth of converted image and LittlevGL should match. And depending on the output format of the camera (BMP, JPG, PNG etc) you should use an external library to get the raw pixel data.

Some improvements clearly need to be done in this field because the camera image displaying and video playing seems to be a hot topic now. A lot people already asked it.
I think - as the first step - a good decoder interface would be required to handle different file types. For example to make possible to set a PNG or JPEG decoder callback. In this case LittlevGL could decode the images when it is required. It would be slower but more convenient.

Masi2017, 2018-04-27:
@kisvegabor I tried to use Image converter and converted a jpg file to binary888, but the file extension is .bin.c , is that correct?

kisvegabor, 2018-04-30:
Hi,

The ".bin.c" was incorrect. I fixed it.

To use a pointer you need to set up an image description manually using the address of the binary data (`MY_IMG_DATA_ADRESS` in the example):
```c
    static lv_img_t my_img_data_1;
    my_img_data_1.header.w = 100;    /*Width in pixels count*/
    my_img_data_1.header.h = 150;    /*Height in pixels count*/
    my_img_data_1.header.alpha_byte = 0;    /*Pixel level opacity [0/1] (Was an option in the font converter tool)*/
    my_img_1.header.chroma_keyed = 0;    /*Chroma keyed image [0/1] (Was an option in the font converter tool)*/
    my_img_data_1.header.format = LV_IMG_FORMAT_INTERNAL_RAW;    /*It's like an internal image because it will use pointer access to the image*/
    my_img_data_1.pixel_map = MY_IMG_DATA_ADRESS + 4;    /*Address of the binary data (+4 to skip the description in the beginning)*/
    
    lv_obj_t * img_obj = lv_img_create(lv_scr_act(), NULL);
    lv_img_set_src(img_obj, my_img_1);
```

However didn't tested it but it should work.

Masi2017, 2018-05-01:
Hi
Thank you very much I tested and it worked.

kisvegabor, 2018-05-02:
Hi,

Great! Thank you for the feedback!

kisvegabor, 2018-07-15:
As lastly suggested solution is tested and worked I close this issue.

closed 2018-07-15 by kisvegabor

Masi2017, 2018-09-27:
Hi @kisvegabor 
I updated littlvgl to v5.2 so now I'm getting an error using the above image description , would you please let me know how can use the binary files now?

kisvegabor, 2018-09-28:
Here it is:

```c
    static lv_img_dsc_t my_img_data_1;
    my_img_data_1.header.w = 100;    /*Width in pixels count*/
    my_img_data_1.header.h = 150;    /*Height in pixels count*/
    my_img_data_1.header.cf = LV_IMG_CF_TRUE_COLOR;
    my_img_data_1.header.always_zero = 0;
    my_img_data_1.data_size = binary_size_in_bytes; /*Not important in this case. Can be set to 0*/
    my_img_data_1.data = MY_IMG_DATA_ADRESS + 4;    /*Address of the binary data (+4 to skip the description in the beginning)*/

    lv_obj_t * img_obj = lv_img_create(lv_scr_act(), NULL);
    lv_img_set_src(img_obj, &my_img_data_1);
```

Besides you should use the image convert for `dev-5.2`: http://www.test.littlevgl.com/image-to-c-array

Masi2017, 2018-10-16:
@kisvegabor Thank you very much. Another question, Does the image converter tools work for older lvgl version ( 5.2 beta for example ) ? I noticed that you have updated the master branch , last version is 5.3, right and I am guessing that the converter tools only works with new lvgl version.

kisvegabor, 2018-10-16:
Uh, The version number was set to v5.3 accidentally. Thank you for pointing this out! The current version is v5.2 and I fixed the version number.
So the converter tool is compatible with the current version!

Masi2017, 2018-10-17:
@kisvegabor Thank you. For some reasons I need the older version of the converter , Is there any way to have access to that ?

kisvegabor, 2018-10-17:
They are available here: https://github.com/littlevgl/lv_utils

Masi2017, 2018-10-17:
Thank you. I copied the "beta" version and then run the command but I got an error:

[image]

kisvegabor, 2018-10-18:
It's PHP not Python

Masi2017, 2018-10-18:
Sorry I got confused, the one which is in the master branch seems to be the latest version of the converter while I am looking for the older version which works with lvgl version 5.1 and 5.2 beta. Maybe I am wrong, I would appreciate that if you let me know which one should be used for older lvgl version? ( beat , master or 5.2 ? ) since I tried beta which is python and I got an erro ( first screenshot) .

kisvegabor, 2018-10-19:
I added the old version here: https://littlevgl.com/image-to-c-array-old

Masi2017, 2018-10-22:
@kisvegabor Thank you very much.