lvgl/lvgl #152: Fonts for monochrome displays

drabdominik, 2018-03-08:
Hi, I'm following up on the discussion about fonts here: #125. For use on monochrome LCDs we need pixel perfect fonts that don't produce artefacts in small sizes. @upbeat27 provided modified version of [unscii-8-thin](http://pelulamu.net/unscii/) font, which can be converted to perfect 1bpp bitmap using online converter. I tried to edit some other fonts but with little success :D. I found, however, many bitmap fonts in .bdf format that can easily be converted to source code. Here's preview of fonts I got working so far:

[image]

Second and third ones are converted using online converter from vector fonts([unscii](http://pelulamu.net/unscii/) and [synchronizer](http://www.fontspace.com/total-fontgeek-dtf-ltd/synchronizer-nbp)). The rest I converted from fonts available in [u8g2](https://github.com/olikraus/u8g2) library, that has hundreds of bitmap fonts for graphic LCDs. Library is open source, but fonts are available under different licences (see [here](https://github.com/olikraus/u8g2/wiki/fntgrp)). Here is another font with about 360 symbols on 12x12 grid ([source](https://github.com/stark/siji)):

[image][image]

I think there could be some collection of such fonts available in littlevGL for people using low resolution display. Or maybe add support for bitmap fonts in online converter ? Anyway, here are converted fonts: 
[fonts.zip](https://github.com/littlevgl/lvgl/files/1793640/fonts.zip)

kisvegabor, 2018-03-09:
Hi,

It looks very well, thank you very much!

I wonder how did you forced the converter to generate fix width letters. Now it should store the in the smallest possible place and trim every empty columns on the left ad right.  (for example in *f_6x10.c*)

I think it would be great to add at least one pixel font (e.g. in 7 or 8 px height) with ASCII and Symbol characters as built in font. It would make much easier to start the development  on monochrome displays. What do you think?

drabdominik, 2018-03-09:
I actually made my own converter for .bdf fonts. First i tried to convert them into vector fonts and use your online converter but it didn't work. Fonts I used have glyphs with constant size (e.g. 6x10), that's why all characters are same width.

Yes, it would nice to have some font in library as starting point. Unscii is probably good candidate as it is public domain. The symbol font (siji) however, is GPL and as far as i understand, it's not compatible with MIT licence. So maybe if someone could find similar font with different licence it could be used.

kisvegabor, 2018-03-12:
Okay, if you find it useful I will share the php files in [lv_utils](https://github.com/littlevgl/lv_utils) repo.