lvgl/lvgl #153: Online convert tools

BenQoo, 2018-03-09:
The image and font covert tools have been ported to website at v5.1.
Would you provide the corresponding offline tools in the future like the older versions?
Because some images and fonts are properties which are not suitable to upload to website.

kisvegabor, 2018-03-09:
Hi,

I see your problem.
However I'm not planning to maintain an offline tool simultaneously I can add a note to website which states: "The images and fonts are not stored on the website". And really the uploaded images and fonts exist only during the converting process.

Besides if someone would like to create the offline version of the Font and/or Image converter I could provide the PHP scripts as a reference.

hreichert, 2018-03-09:
@kisvegabor It would be great if you publish the PHP scripts in a git repository, so someone has also the chance to run them locally.

WangZhihua1998, 2018-03-09:
@kisvegabor If will be helpful if you are willing to release a php script since we can manage to run a php script on *nix systems easily.

kisvegabor, 2018-03-12:
Okay, if it helps I will publish the php scripts in [lv_utils](https://github.com/littlevgl/lv_utils) repo!

kisvegabor, 2018-03-12:
I uploaded the scripts to lv_utils. They are not so well formatted like LittlevGL. I didn't ment to publish them :)

kisvegabor, 2018-03-13:
I updated the scripts and readme to make offline conversion easier.
Please have a look at it!

kisvegabor, 2018-05-17:
As the offline tools are available now I close this issue.
If you have any remarks feel free to reopen it!

closed 2018-05-17 by kisvegabor