lvgl/lvgl #156: C++ compatibility

memen45, 2018-03-09:
Is there a way to get the library working in a C++ project? The rest of my project is in C++ and when I switch the compiler to g++ instead of gcc, I run into all kinds of ' invalid conversions' that c++ does not allow.

The simplest way could be using a gcc compiler separately for the library, however, this requires a lot of maintenance in the makefile. Or am I missing some solutions?

Are there any plans yet to either make it fully c++ compliant or - maybe forked - even c++ based?

memen45, 2018-03-10:
I have to correct my previous post partially. The problems with c++ compliancy are not in this library, but in the lv_drivers repository. 
With the current library it is only possible to have one display at a time, while c++ could allow for multiple ones easily. Also memory management could be performed safer in c++ as well as proper de-initialization of the display and it could be easier to implement. Therefore the question remains: Are there any plans yet to make it c++ based?

Zaltora, 2018-03-11:
I don't think lvgl or a part will be ported in c++. Maybe some extern "C" are missing and need to be fixed to be fully supported in c++. Anyway, you need compile with both compilers gcc/g++ but i don' no why you need a lot of maintenance ? 
I work with a SDK that support C/C++ compilation Maybe it will help you: [common.mk](https://github.com/SuperHouse/esp-open-rtos/blob/master/common.mk)

Multi-display feature are not ready yet in lvgl (it is planning) but drivers are currently rewrote to support this with struct pointer (object). We also work on a HAL interfaces to make easily portable the drivers.

kisvegabor, 2018-03-12:
I tested LittelvGL in C++ environment when I built an Arduino library from it. Then I've solved a lot of issues like adding `extern "C" {` to the header files.

However these update are really missing from the *lv_driver* headers. 
Can you try to updates the drivers you use according to 'lv_misc/lv_templ.h'? I hope it will solve the issue. 

As @Zaltora said we are working on a new driver API which will be ready to support multiple displays. However unfortunately  we can't use C++ "features" in LittlevGL because C compatibility must be kept.

memen45, 2018-03-12:
Yes, I can do that. Wasn't too much work. To which branch should I make a pull request?

C compatibility could be maintained using a C-interface (a C-header file that can be included from elsewhere, combined with a cpp source file implementing the c-style functions in terms of the C++ library. For example a driver function for the ssd1963 like this:

c-style header (e.g. ssd1963.h):
```c
void ssd1963_init(void);
``` 

c++ style source file - but with c-style functions
```c++
#include "ssd1963.h"         /* c-style header   */
#include "ssd1963.hpp"       /* c++ style header */
#include "lvgl/lvglc++.hpp"  /* c++ style header */

SSD1963_Driver g_disp_driver;

void ssd1963_init(void) {
    g_disp_driver.init();
}
```

However, I understand that translating a whole library to C++ will be a lot of work, but I was just curious if anyone was working on it!

kisvegabor, 2018-03-13:
According to my experiences on C/C++ compatibility  `extern "C" {` define in the headers solve the issues. Of course it won't provide C++ styled code, but make LittlevGL possible to compile.  :)

To answer your question: currently nobody is working on it and I don't think it would be reasonable to  maintain a C and C++ styled code if the C styled works well in C++ too. 
@Zaltora  is doing a great work on reworking the driver's API and I'm sure he can add the extern "C" {` defines to the header to provide the compatibility.

If you have other thought, please let us know. :)

kisvegabor, 2018-03-13:
@memen45 I merged you Pull Request. Thank you very much!
Can you compile LittelvGL in C++ environment right now?

memen45, 2018-03-13:
Yes, I can compile it! Actually looks nice as well and I really like that it can be simulated on pc as well. It makes development a lot easier. I'll write my own drivers anyways, since I need them to be asynchronous, but I'm glad I could help!

kisvegabor, 2018-03-13:
Great! :)
I also found development on PC much easier and faster. 
If you face some LittlevGL related issue while writing the drivers just open an issue in [lv_drivers](https://github.com/littlevgl/lv_drivers) repository. 

As the issue is solved I close this thread.

closed 2018-03-13 by kisvegabor