lvgl/lvgl #157: Unexpected font coversion bitmap

BenQoo, 2018-03-12:
There are some issue about Font online converter

*  some glyphs could be cut.

```C
  /*Unicode: U+21e7 (⇧) , Width: 17 */
  0x00, 0x00, 0x00,  //................. 
  0x00, 0x80, 0x00,  //........%........ 
  0x01, 0x40, 0x00,  //.......@+@....... 
  0x02, 0x20, 0x00,  //......@...@...... 
  0x04, 0x10, 0x00,  //.....@....+@..... 
  0x08, 0x08, 0x00,  //....@+.....+@.... 
  0x10, 0x04, 0x00,  //...@+.......+@... 
  0x20, 0x02, 0x00,  //..@+.........+@.. 
  0x7c, 0x1f, 0x00,  //.%@@@@.....@@@@%. 
  0x04, 0x10, 0x00,  //.....@.....@..... 
  0x04, 0x10, 0x00,  //.....@.....@..... 
  0x04, 0x10, 0x00,  //.....@.....@..... 
  0x04, 0x10, 0x00,  //.....@.....@..... 
  0x04, 0x10, 0x00,  //.....@.....@..... 
  0x04, 0x10, 0x00,  //.....@.....@..... 
  0x04, 0x10, 0x00,  //.....@.....@..... 
  0x04, 0x10, 0x00,  //.....@.....@..... 
  0x04, 0x10, 0x00,  //.....@.....@..... 
  0x04, 0x10, 0x00,  //.....@.....@..... 
  0x04, 0x10, 0x00,  //.....@.....@..... 

  /*Unicode: U+2b6e (⭮) , Width: 21 */
  0x00, 0x00, 0x00,  //..................... 
  0x0f, 0x00, 0x00,  //...+%%@%............. 
  0x3f, 0x00, 0x00,  //.+@@@@@%............. 
  0x1e, 0x00, 0x00,  //..+@@@@+............. 
  0x1e, 0x00, 0x00,  //...@@@@.............. 
  0x3e, 0x00, 0x60,  //..@@@@@..........@@.. 
  0x72, 0x00, 0x70,  //.%@@.+@..........@@%. 
  0x60, 0x00, 0x30,  //.@@...............@@. 
  0x60, 0x00, 0x30,  //+@%...............%@+ 
  0xc0, 0x00, 0x18,  //%@+...............+@% 
  0xc0, 0x00, 0x18,  //@@.................@@ 
  0xc0, 0x00, 0x18,  //@@.................@@ 
  0xc0, 0x00, 0x18,  //@@.................@@ 
  0xc0, 0x00, 0x18,  //@@.................@% 
  0xe0, 0x00, 0x30,  //%@+...............%@+ 
  0x60, 0x00, 0x30,  //.@@...............@@. 
  0x70, 0x00, 0x70,  //.%@%.............%@%. 
  0x30, 0x00, 0x60,  //..@@+...........+@@.. 
  0x1c, 0x01, 0xc0,  //..+@@%.........%@@+.. 
  0x0f, 0x07, 0x80,  //...+@@@%.....%@@@+... 
```

* Some glyphs will be output as one or two tofu.

```C
  /*Unicode: U+1f5d8 (🗘) , Width: 13 */
  0x7f, 0xf0,  //+@@@@@@@@@@@. 
  0x40, 0x30,  //+@++++++++%@. 
  0x40, 0x10,  //+@........+@. 
  0x40, 0x10,  //+@........+@. 
  0x40, 0x10,  //+@........+@. 
  0x40, 0x10,  //+@........+@. 
  0x40, 0x10,  //+@........+@. 
  0x40, 0x10,  //+@........+@. 
  0x40, 0x10,  //+@........+@. 
  0x40, 0x10,  //+@........+@. 
  0x40, 0x10,  //+@........+@. 
  0x40, 0x10,  //+@........+@. 
  0x40, 0x10,  //+@........+@. 
  0x40, 0x10,  //+@........+@. 
  0x40, 0x10,  //+@........+@. 
  0x40, 0x10,  //+@........+@. 
  0x40, 0x10,  //+@........+@. 
  0x40, 0x10,  //+@........+@. 
  0x40, 0x30,  //+@++++++++%@. 
  0x7f, 0xf0,  //+@@@@@@@@@@@. 

  /*Unicode: U+1f82c (🠬) , Width: 37 */
  0x7f, 0xf0, 0x00, 0x7f, 0xf0,  //+@@@@@@@@@@@............+@@@@@@@@@@@. 
  0x40, 0x30, 0x00, 0x40, 0x30,  //+@++++++++%@............+@++++++++%@. 
  0x40, 0x10, 0x00, 0x40, 0x10,  //+@........+@............+@........+@. 
  0x40, 0x10, 0x00, 0x40, 0x10,  //+@........+@............+@........+@. 
  0x40, 0x10, 0x00, 0x40, 0x10,  //+@........+@............+@........+@. 
  0x40, 0x10, 0x00, 0x40, 0x10,  //+@........+@............+@........+@. 
  0x40, 0x10, 0x00, 0x40, 0x10,  //+@........+@............+@........+@. 
  0x40, 0x10, 0x00, 0x40, 0x10,  //+@........+@............+@........+@. 
  0x40, 0x10, 0x00, 0x40, 0x10,  //+@........+@............+@........+@. 
  0x40, 0x10, 0x00, 0x40, 0x10,  //+@........+@............+@........+@. 
  0x40, 0x10, 0x00, 0x40, 0x10,  //+@........+@............+@........+@. 
  0x40, 0x10, 0x00, 0x40, 0x10,  //+@........+@............+@........+@. 
  0x40, 0x10, 0x00, 0x40, 0x10,  //+@........+@............+@........+@. 
  0x40, 0x10, 0x00, 0x40, 0x10,  //+@........+@............+@........+@. 
  0x40, 0x10, 0x00, 0x40, 0x10,  //+@........+@............+@........+@. 
  0x40, 0x10, 0x00, 0x40, 0x10,  //+@........+@............+@........+@. 
  0x40, 0x10, 0x00, 0x40, 0x10,  //+@........+@............+@........+@. 
  0x40, 0x10, 0x00, 0x40, 0x10,  //+@........+@............+@........+@. 
  0x40, 0x30, 0x00, 0x40, 0x30,  //+@++++++++%@............+@++++++++%@. 
  0x7f, 0xf0, 0x00, 0x7f, 0xf0,  //+@@@@@@@@@@@............+@@@@@@@@@@@. 
```

About the above example could be referred as following:

Font:
[Noto Sans Symbols2](https://www.google.com/get/noto/)

Online page configurations:

| Field    | value                                  |
| -------- | -------------------------------------- |
| TTF file | NotoSansSymbols2-Regular.ttf         |
| Name     | lv_font_notosans_symbols2_regular_18 |
| Height   | 18                                   |
| Bpp      | 4                                    |
| Range    | 8678 ~ 129095                      |
| List     | ⇦⇧⇨⇩⌨⌫⏴⏵⏶⏷␣▲△▶▷▼▽◀◁☐☑☒⛌⬅⬆⬇➡⭮⭯⮐⮜⮝⮞⮟⮹⮽⯅⯆⯇⯈🗘🗙🞀🞁🞂🞃🠄🠅🠆🠇🠈🠉🠊🠋🠘🠙🠚🠛🠜🠝🠞🠟🠬🠭🠮🠯🠰🠱🠲🠳🡄🡅🡆🡇 |
| Built-in | checked                                |

Convert output file: 
[lv_font_notosans_symbols2_regular_18.zip](https://github.com/littlevgl/lvgl/files/1801791/lv_font_notosans_symbols2_regular_18.zip)

kisvegabor, 2018-03-12:
Hi,

Thank you for the detailed explanation!

**The trimmed glyps**
The font converter needs to adjust the height of the fonts to fit to the desired height [px]. But unfortunately it could be incorrect with some special characters. To solve it I can add two fields to the converter's form: 
1. Height scale[%]
2. Base line offset[px]
It needs some extra work from the user to "play" with the values but makes possible to fix any issues in the future.

**Got rectangle(s)**
I examined this issue deeply and found the character renderer gets correct input but creates wrong result.
I loaded this font into [BMFont](http://www.angelcode.com/products/bmfont/) but got even worse result. For example the U+1f82c (🠬) character was not listed at all. 
However in FontForge all the arrows are displayed correctly. I have no ide right now what could cause this issue.

kisvegabor, 2018-06-22:
I added the **Height scale** and **base line offset** features to the font converter to enable fine tunig of these attributes.

BenQoo, 2018-07-04:
Hi,
About the "Got rectangle(s)" issue, is it be solved too?

kisvegabor, 2018-07-06:
Meanwhile, I knew the Font forge software and checked the Noto Sans Symbol Medium font with that. 
I found that the characters which were converted to rectangles are not present in the font.
Look the attached screenshot. It shows that there is no symbol to 🠟 (0x1f81f).

[image]

BenQoo, 2018-07-09:
Please check the font `NotoSansSymbols2-Regular.ttf`.
[image]

kisvegabor, 2018-07-10:
Oh, sorry, you really wrote it earlier.

So I debugged it and I found that the used PHP function (`imagettftext`) gives rectangles for values above 0xFFFF. I tried to convert the font to `otf` but it didn't help.

I need to try other fonts to see it's really a bug in the PHP function or it's maybe related to the font somehow.

kisvegabor, 2018-08-09:
As the PHP function seems buggy I have no idea how to fix it. I lack of any reasonable idea I close this issue. However, if you find something which might help, please comment here.

closed 2018-08-09 by kisvegabor