lvgl/lvgl #161: Clean of list obj breaks object

BenArtes, 2018-03-14:
I'm trying to update the items in a list object and have run into a problem.

My Plan was to clean the list object to delete all the previously attached items in the list, then add / re-add the items I wish to be in the new list.

```c
    lv_obj_clean(list);

    for(uint32_t curr_item_num = 0; curr_item_num < NUM_ITEMS; curr_item_num++)
    {
        const item_t *const curr_item = &items[curr_item_num];
        if(curr_item->item_valid)
        {
            lv_list_add(select_list, NULL, curr_item->item_name, item_selected);
        }
    }
```

The above causes the deletion of `ext->scrl` which (due to the particular ordering of my code) causes the new item to be created at the same address as the previously allocated `ext->scrl` and eventually `lv_obj_set_parent(tmp, ext->scrl);` in `lv_page_signal` executes with `tmp` and `ext->scrl` being equal causing an infinite loop.

* Is there a better way to clear the items in a list without breaking it?  
* Should there be a function for removing individual items from a list and / or clearing a list entirely?
* What other objects have similar problems and what could there solution be?

BenArtes, 2018-03-14:
I see https://github.com/littlevgl/lvgl/issues/25 but am trying not to keep pointers around to every object created (as I have to carefully track and clear / NULL them once I delete them etc).

BenArtes, 2018-03-14:
It turns out I was also running into this issue when trying to clear objects from a tabview page.  I `clean`'d the page and deleted the scrl.

kisvegabor, 2018-03-14:
Hi,

It's really not clear now. So it's a very good idea to add a dedicated function.

As the page contains the scrollable object (`ext->scrl`) and the list elements are on the scrollable you need to clean the scrollable object. 

```c
    lv_obj_t * scrl = lv_page_get_scrl(my_list);
    lv_obj_clean(scrl);
```

All objects are affected which are delivered from `lv_page` and can be used to add other object to them:
- lv_page
- lv_list
- lv_tabview (the created tabs are pages)
- lv_win (the content area is a page)

It would be very good to add a clean function should to all these objects too. Like: `lv_win_clean()`

In my opinion because the infinite loop in `lv_obj_set_parent()` is created only in special cases we don't need prepare for it. According to my experiences situations like this usually cause hard fault.

BenArtes, 2018-03-14:
Okay, I'll work on that and send a PR to dev-5.2

It might be a little confusing that some objects use `lv_obj_clean` and others use `lv_objname_clean`.  Maybe every obj-type should have lv_objname_clean implemented?

kisvegabor, 2018-03-14:
Awesome, thank you in advance!

Good question. Yes, it's maybe confusing. For a while I was think about wrap all `lv_obj_..` functions to every object type. For example: `lv_btn_set_pos()`. In most cases it's just a name change but in some cases (like this one) some modifications are required. With `inline` functions it doesn't mean extra function call and extra memory. However it's a lot more work. What do think about this idea?

BenArtes, 2018-03-14:
Something I hadn't thought of:

If all functions use specific object-type functions (`lv_btn_set_pos()`) instead of the generic object functions (`lv_obj_set_pos()`) refactoring is more work as every function must be changed.  Currently, in many cases, only the `lv_*_create` function needs to be changed.  (Example: changing container to page).  Not sure how much this matters.

If memory wasn't a constraint, the object functions could be stored as function pointers in the object, but I think that would take too much space.

It might be possible to use macros to automatically select the correct function, but that would likely make debugging more difficult (if the macro fails) and include run-time penalty.

I think this should be given more thought and might be a good candidate for a 6.0 change in case it changes / breaks the API.

For now I'll just implement object specific clean function for:
* lv_page
* lv_list
* lv_tabview
* lv_win

kisvegabor, 2018-03-14:
Yes, it's a lot of work but there will be very similar function which can be produced with search+replace. E.g. `lv_btn_set_pos()` `lv_list_set_pos()`, `lv_ta_set_pos()` etc. 
As this changes won't break API compatibility just serve as an extension I think it can go to v5.2 as well. 
But it definitely  needs a test. I'm not sure it so easy as it looks :) I also vote for function instead of macro exactly because of the reasons you mentioned.

Anyway the updating of the 4 affected functions would be perfect now! :clap:

BenArtes, 2018-03-16:
It looks like ddlist, roller, and ta should also have new clean functions implemented so that their scrl object doesn't get deleted.  Is this correct?

kisvegabor, 2018-03-19:
I didn't list these objects because they are not used to add "random" content to them. For example there is no reason to delete the label of a text area. What do you think?

BenArtes, 2018-03-19:
I think you listed the correct objects; just wanted to double-check.

closed 2018-03-19 by BenArtes