lvgl/lvgl #162: slider action on enter

Masi2017, 2018-03-14:


kisvegabor, 2018-03-14:
Hi,

The `lv_slider_set_action(slider, action);` function set an action for slider and the action will be called on every LEFT/RIGHT step. So no action called when you press ENTER.

Do you think it would be better to call the action only on ENTER?

Masi2017, 2018-03-14:
@kisvegabor I think it's good if we have an action on Enter as well otherwise how we can close the slider or move to the other object on the screen ?

kisvegabor, 2018-03-14:
If you have NEXT/PREV buttons then you can use them. Or you don't have these buttons?

Masi2017, 2018-03-14:
oh NO actually , I have 5 buttons only (Right,Left,Up,Down and Enter), I think I  need to add Next and Prev on long press Right and left or do something else.

kisvegabor, 2018-03-16:
I see. However it's a good idea to rework the KEY actions to make possible to operate without NEXT/PREV buttons. 
If I understood well you expect that user will press and ENTER or ESC and the action will be called. And in the action you focus on the next object manually.

Masi2017, 2018-03-16:
@kisvegabor  Yes exactly.

kisvegabor, 2018-03-17:
Okay, I also found approach very useful. It would be great to support this usage as well. 
In v5.2 we can add this updates. Please keep sending your remarks! :)

kisvegabor, 2018-07-25:
Hi @Masi2017 

Please see this issue: https://github.com/littlevgl/lvgl/issues/261

A new feature is introduced to support keyboards with few keys or even encoders (only left, right, push).
It should be the solution for your issue too.

As of now,  the mentioned new feature is the prefered way to handle a few keys I close this issue. If you have any remarks please add them to the topic above.

Thank you!

closed 2018-07-25 by kisvegabor