lvgl/lvgl #163: no space, what is wrong ?

yaoyutaoTom, 2018-03-15:
mcu is stm32f103vc, External display controller is ssd1963. 7 '' tft.  

Lcd\Lcd.axf: Error: L6406E: No space in execution regions with .ANY selector matching lv_mem.o(.data).
Lcd\Lcd.axf: Error: L6406E: No space in execution regions with .ANY selector matching lv_theme.o(.data).
Lcd\Lcd.axf: Error: L6406E: No space in execution regions with .ANY selector matching lv_draw.o(.data).

 when I compile the code in mdk.

kisvegabor, 2018-03-15:
Hi,
STM32F103 is very limited in memory so you need to be very careful what is enabled an what is not.

Please send us your `lv_conf.h` to see your configuration.

kisvegabor, 2018-05-17:
This issue was inactive for 2 moths so I close it now.
If you have any update in this field feel free to reopen it!

closed 2018-05-17 by kisvegabor