lvgl/lvgl #164: 18bits color

Masi2017, 2018-03-15:
I found a library for ILI9341,  I want to port that into lvgl library, but since the code is written for 18bits color depth, I have to use that but in lvgl there 18bits has not been defined, any idea what should I do?

ramonschepers, 2018-03-15:
i would suggest using 24bit color depth, and ditch the upper 2 bits of each color when pushing to screen (so you reduce total colors by 6 bits, resulting in 18bit instead of 24bit), also, keep in mind you need to port the 18bit screendraw code from the library you found yourself, note if you're doing this over SPI, this requires 3 writes per pixel (3x 8bits, of which 2 empty) instead of 16bits, and will make drawing time around 150% of 16bit. (example if 16bit draw takes 25ms, a 18bit will take around 37-38ms for a draw.)

Masi2017, 2018-03-15:
Gffd

kisvegabor, 2018-03-16:
Hi Masi,

What about modifying the code to  init. code of ILI9341 to work with 16 bit colors. You lose only 2 bits but you need to send only 2 bytes? 

In the first run the whole screen will be drawn. Doesn't matter you create anything or not. The library will draw the screen in the beginning. However later if you modify for example a label's text only the label will be redrawn.

kisvegabor, 2018-03-23:
@Zaltora RGB666 is not a common protocol. Do you mean RGB888?

kisvegabor, 2018-03-27:
Oh, really my bad. It's the 18 bit interface.
Now there is no way to add unique colors because in some cases they need special handling. However you can modify the `lv_color24_t` structure. In this case `LV_COLOR_DEPTH` also need to be updated.

Masi2017, 2018-03-29:
@Zaltora Thanks I have the driver now and I ported lvgl library too but then I have to switch on ILI9488. However, I used "loboris driver" and now it works fine.

kisvegabor, 2018-07-16:
Support of 18-bit color format can happen with the next planned feature called "Application managed VDB": https://github.com/littlevgl/lvgl/issues/300

It will enable to manage the VDB (write rad pixels) by custom, user-defined functions. If you have comments and remarks about this topic, please use the new thread!

closed 2018-07-16 by kisvegabor