lvgl/lvgl #165: How does padding affect `lv_obj_get_width` etc?

BenArtes, 2018-03-15:
I'm having some difficulty laying out my objects.  I find that, when setting width and height, I frequently have to add / subtract small factors to account for padding.

Am I doing something wrong, or are there improvements to how padding, width, and height are handled that we can make?

## Example:
> I have a tabview container the size of the whole screen.  I want to evenly divide the page / content area into two child containers that will each have additional child containers for layout.
```
-----------------------
| ---------  -------- |
| |       | |       | |
| |       | |       | |
| ---------  -------- |
-----------------------
```

```c
#define PADDING_OFFSET 8
    // parent = lv_tabview_add_tab(tabview, "Tab");
    lv_page_set_sb_mode(parent, LV_SB_MODE_OFF);
    lv_page_set_scrl_layout(parent, LV_LAYOUT_ROW_T);

    {
        lv_obj_t *details_cont = lv_cont_create(parent, NULL);
        lv_cont_set_layout(details_cont, LV_LAYOUT_COL_L);
        lv_cont_set_fit(details_cont, false, true);
        lv_obj_set_width(details_cont, (lv_obj_get_width(parent) / 2) - PADDING_OFFSET);
        {
            lv_obj_t *child_cont_1 = lv_cont_create(details_cont, NULL);
            lv_cont_set_layout(child_cont_1 , LV_LAYOUT_OFF);
            lv_cont_set_fit(child_cont_1 , false, true);
            lv_obj_set_width(child_cont_1 , lv_obj_get_width(details_cont) - PADDING_OFFSET);

            lv_obj_t *child_label_1 = lv_label_create(child_cont_1 , NULL);
            lv_label_set_text(child_label_1 , "Child 1");
            lv_obj_align(child_label_1 , NULL, LV_ALIGN_IN_LEFT_MID, 4, 0);

            lv_obj_t *child_sw_1 = lv_sw_create(child_cont_1 , NULL);
            lv_sw_set_action(child_sw_1 , child_1_sw_callback);
            lv_obj_align(child_sw_1 , NULL, LV_ALIGN_IN_RIGHT_MID, -4, 0);
        }

        {
            lv_obj_t *child_cont_2 = lv_cont_create(details_cont, NULL);
            lv_cont_set_layout(child_cont_2 , LV_LAYOUT_OFF);
            lv_cont_set_fit(child_cont_2 , false, true);
            lv_obj_set_width(child_cont_2 , lv_obj_get_width(details_cont) - PADDING_OFFSET);

            lv_obj_t *child_label_2 = lv_label_create(child_cont_2 , NULL);
            lv_label_set_text(child_label_2 , "Child 2:");
            lv_obj_align(child_label_2 , NULL, LV_ALIGN_IN_LEFT_MID, 4, 0);

            lv_obj_t* child_ddlist_2 = lv_ddlist_create(child_cont_2 , NULL);
            lv_obj_set_width(child_ddlist_2 , lv_obj_get_width(child_cont_2) - PADDING_OFFSET);
            lv_ddlist_set_hor_fit(child_ddlist_2 , true);

            lv_ddlist_set_options(child_ddlist_2 , "Option 1\nOption 2\n Option 3");
            lv_obj_align(child_ddlist_2 , NULL, LV_ALIGN_IN_RIGHT_MID, -4, 0);
        }
    }

    {
        lv_obj_t *results_cont = lv_cont_create(parent, NULL);
        lv_cont_set_layout(results_cont, LV_LAYOUT_COL_R);
        lv_cont_set_fit(results_cont, false, true);
        lv_obj_set_width(results_cont, (lv_obj_get_width(parent) / 2) - PADDING_OFFSET);
        {
            lv_obj_t *child_cont_1 = lv_cont_create(results_cont, NULL);
            lv_cont_set_layout(child_cont_1 , LV_LAYOUT_PRETTY);
            lv_cont_set_fit(child_cont_1 , false, true);
            lv_obj_set_width(child_cont_1 , lv_obj_get_width(results_cont) - PADDING_OFFSET);

            lv_obj_t *child_label_1 = lv_label_create(child_cont_1 , NULL);
            lv_label_set_text(child_label_1 , "Some Val:");

            lv_obj_t *child_label_2 = lv_label_create(child_cont_1 , NULL);
            lv_label_set_text(child_label_2 , "1024");
        }

        {
            lv_obj_t *child_cont_2 = lv_cont_create(results_cont, NULL);
            lv_cont_set_layout(child_cont_2 , LV_LAYOUT_COL_M);
            lv_cont_set_fit(child_cont_2 , false, true);
            lv_obj_set_width(child_cont_2 , lv_obj_get_width(results_cont) - PADDING_OFFSET);

            lv_obj_t *btn = lv_btn_create(child_cont_2 , NULL);
            lv_btn_set_toggle(btn , false);
            lv_obj_t *btn_label = lv_label_create(btn , NULL);
            lv_label_set_text(btn_label , "Button");
            lv_btn_set_fit(btn_label , true, true);
        }
    }
```

Without Offsets (PADDING_OFFSET = 0):
[image]

With Offsets (PADDING_OFFSET = 8):
[image]

BenArtes, 2018-03-15:
The above images show the following issues:
* The two main Column containers aren't correctly sized / aligned.  Due to padding, setting both containers to the obvious choice of  `lv_obj_get_width(parent) / 2`, after padding, causes the containers to be incorrectly sized, mis-aligned, and spill off the screen.
* Further, setting child containers to `lv_obj_get_width(parent_container)`, after padding etc, also causes them to be too large and spill out of the parent container.
* Not shown, when using the tabview's `ext->content` and `lv_obj_get_height` to create objects within the tabviews content field that are sized to fill the entire content area causes the created objects to be too tall and a either cut off, or the scroll bar active.

kisvegabor, 2018-03-16:
Hi,

The object's width/height always shows their real width/height without any modification. If the object has padding and its content is "moved" then you need to calculate with the padding. Usually I use `style.body.padding.hor` instead of define-ing a value.

I see that is is inconvenient now. In my opinion adding two function to `lv_cont` could solve it:
- `lv_cont_get_width_usable`
- `lv_cont_get_height_usable`  

What do you think? Do you know a better word for *"usable"*?

The improvement should be done for all container like objects (using inline functions to wrap):
- page (and its scrollable)
- list
- window
- tabview
- button

BenArtes, 2018-03-16:
That is exactly what I was thinking.  The name `usable` will work well.

When I have find some time I'll make the changes and send a PR to dev-5.2.

BenArtes, 2018-03-16:
One issue with implementing it this way is that object-type of the parent must be known at the time the child object is created.  I can't think of an easy way around this right now, but it prevents de-coupling the child object's creation:

```c
void create_parent(lv_obj_t *scr)
{
    lv_obj_t*parent = lv_cont_create(scr, NULL);
    // ... Creation of other functions
}

void create_reusable_child(lv_obj_t *parent)
{
    lv_obj_t *reusable_cont = lv_cont_create(parent, NULL);
    lv_cont_set_fit(reusable_cont, false, true);
    lv_obj_set_width(reusable_cont, lv_cont_get_width_usable(parent)); // << Need to know type of parent here
    // ...
}
```

In the above, we can't create a function that creates an object that can be used for different types of parents.

BenArtes, 2018-03-16:
Another Issue:

If we only calculate the usable width based on `style.body.padding.hor` and `style.body.padding.ver` but we want to place multiple items into a container, it won't take into account `style.body.padding.inner`.

Maybe the `lv_cont_get_width_usable` and `lv_cont_get_height_usable` functions should take a parameter for the number of items that will be placed inside the container so we can account for space used by `style.body.padding.inner` as well?

kisvegabor, 2018-03-19:
**Parent type is not known**
A possible solution could be to use the signal functions. Every object type has a unique signal function.  For example a button which is derived from `lv_cont` which is derived from `lv_obj`. When you send a signal firstly `lv_obj` handles it, then `lv_cont` and lastly `lv_btn`. 
We can add two new signals like `LV_SIGNAL_GET_USABLE_WIDTH/HEIGHT` and pass a pointer to an `lv_coord_t` variable to put the result there. 

**Use padding.inner**
To handle this maybe we should typedef a struct with an `lv_cood_t`  and a `children number` elements and pass a pointer to it for the signal function. 

I implemented something similar in `lv_obj_get_type`

BenArtes, 2018-03-19:
**Parent type is not known**
This is what I was trying to get at in our discussions about `lv_obj_clean` vs `lv_page_clean`.  That could have been implemented so that the user calls `lv_obj_clean` on a page and the signal functions are called to correctly clean the page rather than implementing a specific function for cleaning pages.

I think it's worth deciding how you want these features (and similar features in the future) to be implemented for consistency.

My thoughts are:
* Features / functions that exist across all objects should be access with `lv_obj` functions, using signals to implement any object-type specific functionality.
  * This unfortunately implies that many functions, like `clean`, `get_width_usable`, etc may have increased function calls and slightly decreased performance
* Features / functions that only exist for a single object-type should be accessed through that object's functions: 
  * Example: textarea one line is accessed through `lv_ta_set_one_line` because no other object-type has that feature / function
* Features / functions that exist on many object-types but not all object-types should be access through the common ancestor, using signals to implement any object-type specific functionality
  * Example: accessing or edit the Page of Tabview should use `lv_page_*` functions
  * Not sure how practical this is.

I like the above because it increases the re-usability of a lot of code and decreases the number of required changes if a user changes an object's type.

The other option is to, as you'd discussed in Issue #161, implement all functions for all object-types.

Let me know and I'll implement when I get a chance.

kisvegabor, 2018-03-21:
Thank you for the detailed analysis.
I was thinking a lot about it  but  still not clear to me what would be the best.

Now the very common `lv_obj_...` functions (like `lv_obj_set_pos`, `lv_obj_set_parent`, `lv_obj_set_del` etc) send signal to notify the derived types. However for example `lv_obj_set_drag` don't send signal but `lv_win` could use it to make the header dragable instead of the window itself (like the windows on PC).

If we add signals to every `lv_obj_...` functions then we should add to other common functions too (like settings the layout with `lv_cont_set_layout` or `lv_btn_set_layout`) which would lead to enormous number of signals and as you mentioned big performance loss.

The other option is the wrap all ancestor (or just the reasonable) function to the given object type (to result `lv_btn_set_pos`, `lv_btn_set_drag`, `lv_btn_set_layout`) sounds very consistent because you can use only the functions of the given type. It also make possible to extend the functionality instead of just wrap the function (like draging of window).   
It is questionable in this case why should we use signal at all? In my opinion as LittlevGL uses some kind of inheritance (`lv_btn` <- `lv_cont` <- `lv_obj`) to make object reusable the signals are still required. It's an easy way to include are the ancestor behavior to an event and add something new  too. But signals should be general information (like the object is pressed, or coordinates are changed)  too keep them independent from the object type.   
However the function wrapping won't solve the "parent type is not known" issue and makes the things even worse. On the other hand it highly increases the consistency. 

Right now I can't see a reasonable, consistent and efficient solution to use arbitrary parent with the same functions. Maybe we reached the limits of C language. 

Although it's only my current opinion. I'm absolutely open to rework the API or architecture of LittelvGL and I'm very curios about your thought.  So what do you think?

BenArtes, 2018-03-27:
Sorry for the delay, I'm still giving this thought.

I'm currently thinking that instead of (or in addition to) the `*_get_width_usable()` functions, it might be better to add additional layouts so that users don't need to manually worry about the various widths and paddings.  My use of the width functions is so that I have more control of layout and can emulate a more advanced layouts of GUIs like PyQT: inparticular I miss having something like the [GridLayout](http://pyqt.sourceforge.net/Docs/PyQt4/qgridlayout.html).  I'm emulating it by nesting ROW and COL Containers (that really emulate nested [BoxLayouts](http://pyqt.sourceforge.net/Docs/PyQt4/qboxlayout.html)).

More advanced layouts would reduce the need for `*_get_width_usable()` functions as users wouldn't have to manually lay things out.

Because of how I'm using the containers, it might be better to create a *layout* style with 0 padding and inner sizes so that the padding and inner are left to the final containers.  This would also reduce the need for `*_get_width_usable()` function.  If this is a better way to implement layouts it would probably need some documentation and examples for users.

Again, I'll think about this and reply in more details once I have more time.

kisvegabor, 2018-03-28:
Now there isn't  such an advanced layout like Qt's grid. However as you figured out using COL and ROW layout makes it possible to achieve something similar.

kisvegabor, 2018-07-16:
We are planning to rework the `padding` property and add `margin` to be similar to [CSS box model](https://www.w3schools.com/css/css_boxmodel.asp). It will be generally applied by all objects. Read more here: https://github.com/littlevgl/lvgl/issues/302

As it will clarify the issue discussed here I close this thread. If you have ideas or remarks about this topic please add them in the new thread.

Thank you for the contribution! :)

closed 2018-07-16 by kisvegabor