lvgl/lvgl #166: gauge in gauge?

ramonschepers, 2018-03-15:
is it possible to draw a gauge in a gauge, with one inside the other?

for example: the inner gauge measuring value1, and the outer value2

value1 (inner gauge) would need it's text on the inside of the gauge, while value2 (outer gauge) needs it's text on the outside.

a reference pic of what i mean:
[image]

ramonschepers, 2018-03-15:
why would one need this feature? to save space on screen, as a MCU doesn't have a big screen in general, we'll need to think clever about utilizing screen area.

kisvegabor, 2018-03-16:
Hi Ramon,

It seems a reasonable thing to me. To only problem is now the labels always go inside the gauge. If they were go outside then it would be easy to create two on each other (a smaller and a larger).

It is a possible extension for v5.2!

kisvegabor, 2018-10-07:
As there was no activity here for a while I close this issue.

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2018-10-07 by kisvegabor