lvgl/lvgl #167: objects are not created

Masi2017, 2018-03-15:
I am using TFT LCD 240*320, I ported the library in my project but my problem is no matter which object I have created, always just the whole screen will be cleared, I created one lable and set the size and position but nothing happened. my disp_drv.disp_flush function is run and just the whole screen is cleared.I debugged and found that in the  disp_drv.disp_flush function x1,x2,y1,y2 are changed and they are not set according to the object size and position!
I would appreciate if someone can help me.

kisvegabor, 2018-03-16:
Hi,

do you get any result on the screen? If no object are created at all then maybe the screen is not working. Because the default screen color is white and if your screen (display controller?) is not started then you also get a white screen.

Try to modify the default screen color in lv_core/lv_style.c:
```c
    lv_style_scr.body.main_color = LV_COLOR_RED;
    lv_style_scr.body.grad_color = LV_COLOR_RED;
```

Is there any changes?

Masi2017, 2018-03-16:
I debuged and the only thing is happening is that my disp-flash is run and
the hole screen is written  with white color, I tried to change the style
color snd then just the whole screen’s color is changed but no object seems
to be created, it’s like only src screen is loaded and then nothing is
happened. My screen is working since I am anle to write on that if I just
call my disp-flush manually with out using lvgl task.

kisvegabor, 2018-03-16:
I see. I never experienced something like this. 
Please send your "object creating" code. Maybe there is a small in it.

kisvegabor, 2018-03-16:
And please also send your lv_conf.h

Masi2017, 2018-03-16:
Yes sure.

  lv_disp_drv_t disp_drv;                         /*Descriptor of a display driver*/
  lv_disp_drv_init(&disp_drv);                    /*Basic initialization*/

  /*Set up the functions to access to your display*/
  disp_drv.disp_flush = TFT_Flush;            /*Used in buffered mode (LV_VDB_SIZE != 0  in lv_conf.h)*/
  disp_drv.disp_fill = TFT_Fill;              /*Used in unbuffered mode (LV_VDB_SIZE == 0  in lv_conf.h)*/
  disp_drv.disp_map = TFT_Map;                /*Used in unbuffered mode (LV_VDB_SIZE == 0  in lv_conf.h)*/
  //
  /*Finally register the driver*/
  lv_disp_drv_register(&disp_drv);

  lv_obj_t * label1 = lv_label_create(lv_scr_act(), NULL);
  
   static lv_style_t body_style;
  lv_style_copy(&body_style, &lv_style_pretty_color);
  body_style.text.color = TFT_YELLOW1;
  body_style.body.grad_color=TFT_YELLOW1;
  body_style.body.main_color = TFT_YELLOW1;//LV_COLOR_WHITE;

  lv_obj_set_pos(label1,40,60);//20,30
  lv_obj_set_size(label1,80,30);
  lv_obj_set_style(label1, &body_style);

I should add that obj is not created at all and only SCR is created and if I change the style of scr the color will change otherwise changing lable style has no effect since it's not created.

kisvegabor, 2018-03-16:
Thank you!

Please try to create just a  button? Just `lv_btn_create(lv_scr_act(), NULL);` and nothing else.

Just two side notes:
- Label's size set automatically to the text size. (At least in the default long mode)
- You should use  `lv_label_set_style()`. It's the same for the label but not the same for other objects.

Masi2017, 2018-03-16:
[image]

Please see the attached screen shot.  I put a printf function on my Disp-Flush (TFT_Flush) to see the x and y parameters and also how many times it's called.

kisvegabor, 2018-03-16:
It's fine if disp_flush is called multiple time to refresh the whole screen. It seems LV_VDB_SIZE is (LV_HOR_RES * 30) for you.  LittlevGL renders the first 30 line and flush it. After it the next 30 line and flush it and so on.

Then maybe something wrong in `TFT_flush` ? Do you set the coordinates according to the function parameters?

Masi2017, 2018-03-16:
@kisvegabor Yes I know first TFT_flush is run to refresh the whole screen but then to create the object it should be called right? and I think if TFT_flush is not working so the whole screen should not be refreshed.

Masi2017, 2018-03-16:
@kisvegabor Oh thank you I got it  , I thought first screen will be refreshed and then again disp_flush is called to create the object! sorry my bad.

kisvegabor, 2018-03-19:
Hi Masi,

Is there any news?

Masi2017, 2018-03-19:
@kisvegabor Thanks for follow up, The problem was because of TFT_flush since I copied it from another library and then I figured out It just draws the first two rows of the colors throughout the lcd. I made some changes and now it works but it seems that still needs some modification since it has problem on animation object and also keyboard. For example once I create a keyboard I see the following page:
[image]
but then once cursor start blinking the text area is shifted , see the following pic:
[image]

Any idea?

kisvegabor, 2018-03-20:
I suppose you use a 8 bit based protocol (like SPI) to communicate with the display controller. Yes, in this case you need to swap the bytes. There is no good way to change this in LittelvGL becasue with 16 bit colors the Green color would tear to 2 parts:
* `RRRRRGGG GGGBBBBB` Original  
* `GGGBBBBB RRRRRGGG` Swapped

I suggest to do this in the SPI driver. Send the `x+1` byte first and `x` after it (and x += 2 in the end) 

```c
for(x = 0; x < lenght; x+=2) {
  SPISEND = buf[x+1];
  SPISEND = buf[x];
}
```

It seems the original issue is solved so I close it now. If have more questions with "byte swapping" please open a new issue for it and continue the discussion there.

closed 2018-03-20 by kisvegabor

Masi2017, 2018-03-21:
Thanks Gabor, In fact I am using DMA and I had to reversed the buffer's
content before that I pass the buffer to DMA.

Now I want to port lvgl for ILI9488 and I want to use 24 bit color depth
but in lvgl library lv_color_t is 32 bits while I have this structure for
that :

struct {

uint8_t r;
uint8_t g;
uint8_t b ;

} colot_t

would you please help me with this? I'd like to use that  for RGB format
and since I am using SPI so 1 more byte negatively affect the speed.

kisvegabor, 2018-03-22:
Unfortunately there isn't an effective way to get an RGB888 buffer in disp_fush instead of RGBA8888. 
I suggest to move the pixels in the `lv_color_t * buf` in `disp_flush` before sending it to DMA. However the buffer is `const` you can modify it. It won't cause any issue.

Masi2017, 2018-03-22:
Yes I did so, but I thought mayb there is a way to have this format in lvgl
library. Thank you.

kisvegabor, 2018-03-22:
There isn't because dealing with 3 byte numbers instead of 4 byte would cause big performance loss.