lvgl/lvgl #168: Letter space rendering is inconsistent

BenQoo, 2018-03-16:
The rendering letter space could be inconsistent on certain characters.
But, for the most of characters space are fine, so to adjust `letter_space` of `lv_style_t` could not solve this issue properly.

* LVGL version : 5.1
* Font Name : [NotoSans-SemiCondensedMedium.ttf](https://noto-website-2.storage.googleapis.com/pkgs/NotoSans-hinted.zip)
* Height :18 
* [lv_font_notosans_semicondensedmedium_18.zip](https://github.com/littlevgl/lvgl/files/1818188/lv_font_notosans_semicondensedmedium_18.zip)

Please refer to the following screenshots
* Red arrows: space is too tight
* Blue arrows: space is too loose.

Screenshot
[image]
Screenshot with zoom in 600%
[image]

kisvegabor, 2018-03-16:
Hi,

Thank you for the detailed report!

As I see you have used `style.text.letter_space = 0` so it should be no space between letters. If you look carefully the word "current" between 'r' and 'e' there is no space because 'r' has a very light pixel on the right and 'e' has light pixels on the left. 

This issue is typically solved by [Kerning](https://en.wikipedia.org/wiki/Kerning) but it is not implemented in LittlevGL yet. Kerning uses tables to modify the letter space based on latter pairs.

BenQoo, 2018-03-16:
Hi,

Thanks for your explanation, I got it.

> As I see you have used style.text.letter_space = 0 so it should be no space between letters.

You're right. Because set `style.text.letter_space = 1` will make the space too loose, I tried to set it to zero before.
Please let me close this issue.

closed 2018-03-16 by BenQoo

kisvegabor, 2018-03-16:
Your welcome! :)