lvgl/lvgl #170: Change focus on message box

wxd0216, 2018-03-20:
I have five hardware keyboard and remaped to LV_GROUP_KEY_LEFT/RIGHT/UP/DOWN/ENTER.

In this context:
I have two button on a message box, and I want to switch the focus between them with LV_GROUP_KEY_PREV/NEXT. 

lvgl can navigate among the object in a group with LV_GROUP_KEY_PREV/NEXT, but message box is a objects.

Is there any feasible solution?

Thanks

kisvegabor, 2018-03-20:
Hi,

`LV_GROUP_KEY_LEFT/RIGHT` should go to the next button on a message box. 

Is it working there?

wxd0216, 2018-03-21:
Hi,
I found a mistake in my code.
My hardware keyboard looks like this:
     ↑
← OK →
     ↓
Previously, I remped my hardware keyboard to LV_GROUP_KEY_PREV/NEXT/UP/DOWN/ENTER.
After your answer， I changed my key mapping relationship (LV_GROUP_KEY_PREV/NEXT/LEFT/RIGHT/ENTER.). 
And now, they work very well.   But that changed their original function(UP/Down). 

That's not what I expected. So are there any other solutions?

Thanks!

kisvegabor, 2018-03-21:
Hi,
You are right message box should change buttons with UP//DOWN keys too. I pushed an update to support it  into the `beta` branch. Can you test it?

wxd0216, 2018-03-22:
Yes, I've already tested it. It works very well.

Thank you very much for your support.

closed 2018-03-22 by wxd0216

kisvegabor, 2018-03-22:
Great!

Your welcome!