lvgl/lvgl #171: porting littlevgl to stm32f746 diccovery board

liubingyan123, 2018-03-21:
any plane to port littlevgl to stm32f746 diccovery board?

kl-cruz, 2018-03-26:
Hi, I'll try to prepare pull-request soon. Stay tuned!

kl-cruz, 2018-03-26:
Well, it is proof-of-concept solution for this board. I do not use Eclipse as a software for code development so I cannot reproduce makefiles found in the other examples. Instead of that solution I have created Makefile based on the recipe found here: http://mcufreaks.blogspot.com/2013/03/customizable-makefile-for-gcc-arm.html .
Please let me know if You are interested in this code (I can prepare pull-request):
https://github.com/kl-cruz/lv_boards/tree/stm32f7_discovery_integration

kisvegabor, 2018-03-27:
Hi Karol,

Yes I'm very interested in it! I think song Eclipse is easier for a lot of guys so if you don't mind I will "translate" your project to an Eclipse project.

I saw that you are using ChroArt too. It's great! Have you also  tested it ?

kl-cruz, 2018-03-27:
Sure, I tested all the options. As You can see code needs to be refactored a little. I think I will do it today.

kl-cruz, 2018-03-27:
Ok, cleaned a little bit. I am creating pull-request. Eclipse project can be good solution but please look at this line: 
https://github.com/kl-cruz/lv_boards/blob/stm32f7_discovery_integration/stm32f429_discovery_no_os/Release/makefile#L59

I do not know if it will work correctly.

kisvegabor, 2018-03-28:
Thank you!
I will check that line!

kisvegabor, 2018-07-18:
As the project is already merged to lv_project repo I close this issue.

Thank you very much for sharing you project!

closed 2018-07-18 by kisvegabor