lvgl/lvgl #173: Navigation in message box via Keypad

bhavin03, 2018-03-22:
Hi,
 I have implemented a small gui with keypad navigation. Is it possible to select/navigate eg. "Yes" or "No" button in the message box via keypad ? I can navigate in normal gui, when I create objects myself and add those object in a group, but in the message box the buttons are created via text array. So I don't have any pointer on the buttons object to add in a group. Could you please suggest me a solution ?
 
Regards,

kisvegabor, 2018-03-23:
Hi,

You can add the message box to the group and LV_GROUP_KEY_LEFT/DOWN and LV_GROUP_KEY_RIGHT/UP will let you to navigate among the buttons. 

It was an update about the message box navigation here: #170 
So I suggest to use the `beta` branch.

kisvegabor, 2018-07-07:
As this issue wasn't active for months I close it.
If you have further feel free to reopen it!

ErichStyger, 2018-11-26:
I'm facing the same problem with the latest version of the library. I'm able to use a push button (LV_INDEV_TYPE_KEYPAD) which generates LV_GROUP_KEY_NEXT messages, and I'm able to navigate in other controls, e.g. buttons on a screen or lists. But it does *not* work correctly for a message box: navigation just highlights 'Apply' button, does does not move to the 'Close' button.

I'm using the following code:
`static lv_res_t Btn_About_click_action2(struct _lv_obj_t *obj) {
  /* create message box: */
  lv_obj_t * mbox1 = lv_mbox_create(lv_scr_act(), NULL);
  /* set message box text: */
  lv_mbox_set_text(mbox1, "Default message box\n" "with buttons");                    /*Set the text*/
  /*Add two buttons*/
  static const char *btns[] ={"\221Apply", "\221Close", ""}; /*Button description. '\221' lv_btnm like control char*/
  lv_mbox_add_btns(mbox1, btns, mbox_apply_action);
  lv_obj_set_width(mbox1, 120);
  lv_obj_align(mbox1, NULL, LV_ALIGN_IN_TOP_LEFT, 5, 70); /*Align to the corner*/
    gui_group = lv_group_create(); /* create group for touchless navigation */
  lv_indev_set_group(LV_GetInputDevice(), gui_group); /* assign group to input device */
  lv_group_add_obj(gui_group, mbox1);
}
`
It seems to me that the focus does not change to the next item in the dialog box?
Or I'm doing something wrong?

kisvegabor, 2018-11-28:
@ErichStyger 
It's because `lv_mbox` need `LV_GROUP_KEY_LEFT/RIGHT/UP/DOWN` to navigate between the buttons. `LV_GROUP_KEY_NEXT` is used to focus on the next object but the message box is one object and you just want to navigate inside of it. Similarly to a slider where you need `LV_GROUP_KEY_RIGHT` to increase its value.

ErichStyger, 2018-11-28:
Thank you, I have to see how this can be done. I did not find any good example/tutorial/information on this. The challenge is that the UI should be usable with a single push button (is this possible at all?). I have used a short push with LV_GROUP_KEY_NEXT to go through the UI items and a long push (ENTER). Is that feasible at all? Or do I really need more than one button?

kisvegabor, 2018-11-28:
I suggest to use an `LV_INDEV_TYPE_ENCODER`.  https://github.com/littlevgl/lvgl/wiki/Porting#encoder

ErichStyger, 2018-11-28:
Ok, I'll give  `LV_INDEV_TYPE_ENCODER` a try. 
I have now implemented using `LV_GROUP_KEY_LEFT/RIGHT` and the problem is that for a list of objects created with `lv_list_create` it generates a click action? Is that intended? I would expect a click action only for a `LV_GROUP_KEY_ENTER`, but not for `LEFT`, `RIGHT`, `UP `or `DOWN`?

kisvegabor, 2018-11-29:
I tried it in PC simulator with `lv_test_group_1();`

It doesn't generates click action for me. Can you check in teh PC simulator too?

ErichStyger, 2018-11-30:
I appologize: it has have been a problem on my side, it works properly now as far as I can tell.
I'm working now on getting my code implemented with the Encoder type of input, so I suggest you can close this issue. Thank you!

kisvegabor, 2018-11-30:
Glad to hear that you find the issue!

If you face some problems related to the encoder, please open a new issue!

closed 2018-11-30 by kisvegabor