lvgl/lvgl #174: A few nitpicks with ddlist

upbeat27, 2018-03-22:
I've found a few problems with `lv_ddlist`, though some of them technically would be problems with `lv_page`. These are particularly obvious when dealing with small displays. Unscii font is used in these examples.

The first issue is a combination things dealing with the text line spacing and highlighted selection drawing.

The word "Testy" is used because the T includes pixels in the top row of the 8 pixel height font, and the y includes pixels in the bottom row of the 8 pixel height font. So, the word spans the full height of the font.

1. The collapsed ddlist should show only one option regardless of line spacing (a line spacing of 8 here makes it very clear, you wouldn't necessarily see this with a smaller line spacing):
[image]

2. The highlighted option draws more under the text than below the text. It seems to me it should be more balanced and try to draw the highlight such that the text is centered vertically within it.
With line spacing of 8 you can clearly see more pixels below than above:
[image]

With line spacing of 2 there is an extra row below but not above, so the top of the T blends with the black background:
[image]

The other thing is that the scrollbar looks awkward when the ddlist is closed. It would look better if it was centered vertically when collapsed, or disappeared when collapsed.

I know these seem like nitpicky little details but they are pretty significant when dealing with smaller screens with fewer pixels. My intent is only to improve the project 😄

kisvegabor, 2018-03-23:
Hi,

Thank you for the detailed report.
To examine it please send the code and styles you used to create the ddlist.

upbeat27, 2018-03-27:
ddlist code (nothing special I think):
```
	lv_obj_t *dropdown = lv_ddlist_create(lv_scr_act(), NULL);
	lv_ddlist_set_options(dropdown, "Testy\nTest2\nTest3\nTest4\nTest5\nTest6\nTest7\nTest8");
	lv_ddlist_set_fix_height(dropdown, 50);
	lv_ddlist_set_hor_fit(dropdown, false);
	lv_obj_set_width(dropdown, 50);
	lv_ddlist_set_sb_mode(dropdown, LV_SB_MODE_AUTO);
	lv_obj_align(dropdown, NULL, LV_ALIGN_IN_TOP_RIGHT, -10, 10);
```

Theme is the same as that in issue #175 
Play around with `def.text.line_space` in the theme to see how it affects the dropdown.
Font used is unscii.

kisvegabor, 2018-08-09:
Hi,

I know it's a quite old issue but hopefully, you or somebody are still interested in the updates.

I made some test and seems some of the issues are already solved. 
Here is the result with `text.line_space = 4` using the `dev-5.2` branch. The selected rectangle is symmetrical now.
[image]

The intention was to use the default  `LV_SB_MODE_DRAG` which is invisible when the ddlist is closed. However, it is really useful to use `LV_SB_MODE_AUTO`. So I updated the `ddlist` to handle other scrollbar modes too.

kisvegabor, 2018-09-18:
As there was no activity here for a while  I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.

closed 2018-09-18 by kisvegabor