lvgl/lvgl #175: lv_slider bar draws over border when at max value

upbeat27, 2018-03-22:
This is small but easy to fix.
You wouldn't even notice it typically. Add some transparency to the knob though, and you can see it:
[image]

kisvegabor, 2018-03-23:
Hi,
Thank you, I still didn't noticed it. 
Can you send the code creating this slider? (including styles)

upbeat27, 2018-03-27:
Here is a slider, though this code is not really important... I think only the theme really matters for this:
```
	lv_obj_t *slider;
	slider = lv_slider_create(lv_scr_act(), NULL);
	lv_obj_set_size(slider, 100, 14);
	lv_obj_align(slider, NULL, LV_ALIGN_OUT_BOTTOM_LEFT, 0, 0);
	lv_slider_set_range(slider, 0, 10);
```
attached is the modified theme (mono). The colors will look strange, because I am using 8 bit color but then converting that to 4 bit for the OLED.

[lv_theme_mono.zip](https://github.com/littlevgl/lvgl/files/1853286/lv_theme_mono.zip)

kisvegabor, 2018-03-29:
Thank you!

I fixed in the `beta` branch. is it working there?

upbeat27, 2018-03-29:
That fix does work.
I looked into it a bit and I think a different method gives slightly better results, though.

By better results I mean more "balanced" so that when the slider is at (min_value + 1) it looks more like a reflection of (max_value - 1). Currently they look different - of course, it won't necessarily be perfect, because of odd number of pixels and integer math. But it can be better.

It looks like changing it this way:
```
if(slider_w >= slider_h) {
    area_indic.x2 = (int32_t) ((int32_t)(lv_area_get_width(&area_indic) - 1) * (cur_value - min_value)) / (max_value - min_value);
    area_indic.x2 += area_indic.x1;
} else {
    area_indic.y1 = (int32_t) ((int32_t)(lv_area_get_height(&area_indic) - 1) * (cur_value - min_value)) / (max_value - min_value);
    area_indic.y1 = area_indic.y2 - area_indic.y1;
}
```
Gives better results while also fixing the problem.
(That is in the slider object, the same would apply to bar).
Let me know what you think. Hopefully what I'm saying makes sense!

kisvegabor, 2018-04-03:
I think you are right!
Would you like to create  Pull Request If you have already tested the modification?

kisvegabor, 2018-07-07:
Thank you very much!

closed 2018-07-07 by kisvegabor