lvgl/lvgl #176: Add rejection of tab_load to tabview_load_action

BenArtes, 2018-03-27:
In one of my tabview's tabs, users enter information / start a task that must end when they leave the tab.  I use the tabview_load_action to detect if they have started the task; if they have started the task I present a message box ensuring that they want to leave and warning about lost progress with `Ok` and `Cancel` buttons.  If they press `Ok`, the new tab should be loaded, if they press `Cancel` the tabview load should be canceled.

I think it would be a useful feature to modify the tabview callback to have a bool return val indicating whether the new tab should be loaded or not.

Proposed Modification:

```c
typedef bool (*lv_tabview_action_t)(lv_obj_t *, uint16_t);
```

```c
/**
 * Set a new tab
 * @param tabview pointer to Tab view object
 * @param id index of a tab to load
 * @param anim_en true: set with sliding animation; false: set immediately
 */
void lv_tabview_set_tab_act(lv_obj_t * tabview, uint16_t id, bool anim_en)
{
#if USE_LV_ANIMATION == 0
    anim_en = false;
#endif
    lv_tabview_ext_t * ext = lv_obj_get_ext_attr(tabview);
    lv_style_t * style = lv_obj_get_style(ext->content);

    if(id >= ext->tab_cnt) id = ext->tab_cnt - 1;
    bool load_tab = true;
    if(ext->tab_load_action) {
        load_tab = ext->tab_load_action(tabview, id);
    }

    if(load_tab) {
        ext->tab_cur = id;

        lv_coord_t cont_x = -(lv_obj_get_width(tabview) * id + style->body.padding.inner * id + style->body.padding.hor);
        if(ext->anim_time == 0 || anim_en == false) {
            lv_obj_set_x(ext->content, cont_x);
        } else {
#if USE_LV_ANIMATION
            lv_anim_t a;
            a.var = ext->content;
            a.start = lv_obj_get_x(ext->content);
            a.end = cont_x;
            a.fp = (lv_anim_fp_t)lv_obj_set_x;
            a.path = lv_anim_path_linear;
            a.end_cb = NULL;
            a.act_time = 0;
            a.time = ext->anim_time;
            a.playback = 0;
            a.playback_pause = 0;
            a.repeat = 0;
            a.repeat_pause = 0;
            lv_anim_create(&a);
#endif
        }

        /*Move the indicator*/
        lv_coord_t indic_width = lv_obj_get_width(ext->indic);
        lv_style_t * tabs_style = lv_obj_get_style(ext->btns);
        lv_coord_t indic_x = indic_width * id + tabs_style->body.padding.inner * id + tabs_style->body.padding.hor;

        if(ext->anim_time == 0 || anim_en == false ) {
            lv_obj_set_x(ext->indic, indic_x);
        } else {
#if USE_LV_ANIMATION
            lv_anim_t a;
            a.var = ext->indic;
            a.start = lv_obj_get_x(ext->indic);
            a.end = indic_x;
            a.fp = (lv_anim_fp_t)lv_obj_set_x;
            a.path = lv_anim_path_linear;
            a.end_cb = NULL;
            a.act_time = 0;
            a.time = ext->anim_time;
            a.playback = 0;
            a.playback_pause = 0;
            a.repeat = 0;
            a.repeat_pause = 0;
            lv_anim_create(&a);
#endif
        }

        lv_btnm_set_toggle(ext->btns, true, ext->tab_cur);
    }
}
```

Alternately a variation of the `lv_tabview_set_tab_act` function could be added that doesn't trigger the tabview_load_action so that it can be used inside the tabview_load_action.

BenArtes, 2018-03-28:
I don't see it on the dev-v6.0 branch.

kisvegabor, 2018-03-29:
Sorry, it should be added now.

BenArtes, 2018-03-29:
Works well.  I hadn't thought of using the btns signal that way.  Thanks.

kisvegabor, 2018-04-03:
You welcome.  :)

closed 2018-04-03 by kisvegabor