lvgl/lvgl #177: [Question]Will a widget like listview appear in the future version?

chaoking888, 2018-03-29:
Hi, thanks for sharing littlevgl which helps us a lot and gives a way to study.

My question is as shown in the title.  I think a widget like listview is useful in many feilds. Look forward to having a control like listview in the future version. Express my gratitude again!
[image]

kisvegabor, 2018-03-29:
Hi,

I'm happy to hear that LittlevGL was helpful for you. :)

I can see/open the image you uploaded.

chaoking888, 2018-03-29:
I highly appreciate your immediate reply! And sorry about that, because I am a beginner.

May I make bold to ask is there a plan to develop a widget like listview in the future? :-)

kisvegabor, 2018-03-29:
No problem! :)

In #137 we had a discussion about it. It is planned in v5.2. However I can't say an exact date when it will be released.

chaoking888, 2018-03-29:
It's a pity. I didn't see this discussion. I'll close this issue and look forward to the future version.

More hearty thanks to you.

closed 2018-03-29 by chaoking888

kisvegabor, 2018-03-29:
Do you see it on this link? https://github.com/littlevgl/lvgl/issues/137