lvgl/lvgl #180: a bug will be make Segmentation fault

xiaoxinda, 2018-03-31:
This is a headache for me. It has troubled me for days.
The following information from gdb debug:
#0  lv_draw_label (coords=coords@entry=0x77eeee68, mask=mask@entry=0x77eeeeb8, 
    style=style@entry=0x48c404 <style_num_font>, txt=0x0, 
    flag=LV_TXT_FLAG_NONE, offset=offset@entry=0x48fbee <work_mem+922>)
    at lvgl/lv_draw/lv_draw.c:313
#1  0x0040961c in lv_label_design (label=0x48fb90 <work_mem+828>, 
    mask=0x77eeeeb8, mode=<optimized out>) at lvgl/lv_objx/lv_label.c:667
#2  0x00407230 in lv_refr_obj (obj=0x48fb90 <work_mem+828>, 
    mask_ori_p=mask_ori_p@entry=0x77eeef30) at lvgl/lv_core/lv_refr.c:465
#3  0x004072d4 in lv_refr_obj (mask_ori_p=<optimized out>, obj=<optimized out>)
    at lvgl/lv_core/lv_refr.c:443
#4  lv_refr_obj_and_children (top_p=0x48f9d0 <work_mem+380>, 
    mask_p=mask_p@entry=0x77eeef30) at lvgl/lv_core/lv_refr.c:416
#5  0x00407b40 in lv_refr_area_part_vdb (area_p=0x48c480 <inv_buf>)
    at lvgl/lv_core/lv_refr.c:339
#6  lv_refr_area_with_vdb (area_p=0x48c480 <inv_buf>)
    at lvgl/lv_core/lv_refr.c:303
#7  lv_refr_areas () at lvgl/lv_core/lv_refr.c:250
#8  lv_refr_task (param=<optimized out>) at lvgl/lv_core/lv_refr.c:177
#9  0x0040c444 in lv_task_exec (lv_task_p=<optimized out>)
    at lvgl/lv_misc/lv_task.c:271
#10 lv_task_handler () at lvgl/lv_misc/lv_task.c:101

xiaoxinda, 2018-03-31:
I used:
`while(1){
     lv_tick_inc(1);
     lv_task_handler();
     usleep(5000);
}`

xiaoxinda, 2018-04-02:
lv_draw_label (coords=coords@entry=0x77eeee68, mask=mask@entry=0x77eeeeb8, 
    style=style@entry=0x48c404 <style_num_font>, txt=0x0, 
    flag=LV_TXT_FLAG_NONE, offset=offset@entry=0x48fb7e <work_mem+810>)
    at lvgl/lv_draw/lv_draw.c:313
#1  0x0040961c in lv_label_design (label=0x48fb20 <work_mem+716>, 
    mask=0x77eeeeb8, mode=<optimized out>) at lvgl/lv_objx/lv_label.c:667
#2  0x00407230 in lv_refr_obj (obj=0x48fb20 <work_mem+716>, 
    mask_ori_p=mask_ori_p@entry=0x77eeef30) at lvgl/lv_core/lv_refr.c:465
#3  0x004072d4 in lv_refr_obj (mask_ori_p=<optimized out>, obj=<optimized out>)
    at lvgl/lv_core/lv_refr.c:443
#4  lv_refr_obj_and_children (top_p=0x48f9d0 <work_mem+380>, 
    mask_p=mask_p@entry=0x77eeef30) at lvgl/lv_core/lv_refr.c:416
#5  0x00407b40 in lv_refr_area_part_vdb (area_p=0x48c48a <inv_buf+10>)
    at lvgl/lv_core/lv_refr.c:339
#6  lv_refr_area_with_vdb (area_p=0x48c48a <inv_buf+10>)
    at lvgl/lv_core/lv_refr.c:303
#7  lv_refr_areas () at lvgl/lv_core/lv_refr.c:250
#8  lv_refr_task (param=<optimized out>) at lvgl/lv_core/lv_refr.c:177
#9  0x0040c444 in lv_task_exec (lv_task_p=<optimized out>)
    at lvgl/lv_misc/lv_task.c:271
#10 lv_task_handler () at lvgl/lv_misc/lv_task.c:101

kisvegabor, 2018-04-02:
Hi,
Thanks for the detailed report.

1. Are you using v5.1 from `master`?
2. Please send your lv_conf.h
3. Please send your GUI code too

xiaoxinda, 2018-04-02:
i use v5.1  master
it is my lv_conf.h:

`/**
 * @file lv_conf.h
 * 
 */

#ifndef LV_CONF_H
#define LV_CONF_H

/*----------------
 * Dynamic memory
 *----------------*/

/* Memory size which will be used by the library
 * to store the graphical objects and other data */

#define LV_MEM_CUSTOM      0                /*1: use custom malloc/free, 0: use the built-in lv_mem_alloc/lv_mem_free*/
#if LV_MEM_CUSTOM == 0
#define LV_MEM_SIZE    (32U * 1024U)        /*Size memory used by `lv_mem_alloc` in bytes (>= 2kB)*/
#define LV_MEM_ATTR                         /*Complier prefix for big array declaration*/
#define LV_MEM_AUTO_DEFRAG  1               /*Automatically defrag on free*/
#else       /*LV_MEM_CUSTOM*/
#define LV_MEM_CUSTOM_INCLUDE <stdlib.h>   /*Header for the dynamic memory function*/
#define LV_MEM_CUSTOM_ALLOC   malloc       /*Wrapper to malloc*/
#define LV_MEM_CUSTOM_FREE    free         /*Wrapper to free*/
#endif     /*LV_MEM_CUSTOM*/

/*===================
   Graphical settings
 *===================*/

/* Horizontal and vertical resolution of the library.*/
#define LV_HOR_RES          (320)
#define LV_VER_RES          (240)
#define LV_DPI              100

/* Size of VDB (Virtual Display Buffer: the internal graphics buffer).
 * Required for buffered drawing, opacity and anti-aliasing
 * VDB makes the double buffering, you don't need to deal with it!
 * Typical size: ~1/10 screen */
#define LV_VDB_SIZE         (20 * LV_HOR_RES)  /*Size of VDB in pixel count (1/10 screen size is good for first)*/
#define LV_VDB_ADR          0                  /*Place VDB to a specific address (e.g. in external RAM) (0: allocate automatically into RAM)*/

/* Use two Virtual Display buffers (VDB) parallelize rendering and flushing (optional)
 * The flushing should use DMA to write the frame buffer in the background*/
#define LV_VDB_DOUBLE       0       /*1: Enable the use of 2 VDBs*/
#define LV_VDB2_ADR         0       /*Place VDB2 to a specific address (e.g. in external RAM) (0: allocate automatically into RAM)*/

/* Enable anti-aliasing (lines, and radiuses will be smoothed) */
#define LV_ANTIALIAS        1       /*1: Enable anti-aliasing*/

/*Screen refresh settings*/
#define LV_REFR_PERIOD      50    /*Screen refresh period in milliseconds*/
#define LV_INV_FIFO_SIZE    32    /*The average count of objects on a screen */

/*=================
   Misc. setting
 *=================*/

/*Input device settings*/
#define LV_INDEV_READ_PERIOD            50                     /*Input device read period in milliseconds*/
#define LV_INDEV_POINT_MARKER           0                      /*Mark the pressed points*/
#define LV_INDEV_DRAG_LIMIT             10                     /*Drag threshold in pixels */
#define LV_INDEV_DRAG_THROW             20                     /*Drag throw slow-down in [%]. Greater value means faster slow-down */
#define LV_INDEV_LONG_PRESS_TIME        400                    /*Long press time in milliseconds*/
#define LV_INDEV_LONG_PRESS_REP_TIME    100                    /*Repeated trigger period in long press [ms] */

/*Color settings*/
[... 375 lines cut ...]
		break;
	}
}

void gui_set_phonenumber(char *pm) {
	lv_label_set_text(phonenum,pm);
}

void gui_set_agent(char *agent) {
	lv_label_set_text(myagn,agent);
}

void gui_set_time(char *time) {
	lv_label_set_text(mytime,time);
}

void gui_set_number(char *num) {
	if(strlen(num) < 11){
		lv_obj_set_pos(mynum,99,200);
	}else{

		lv_obj_set_pos(mynum,50,200);
	}
	lv_label_set_text(mynum,num);
}`

xiaoxinda, 2018-04-02:
I really need your help.
This problem afflicted me for so many days.

xiaoxinda, 2018-04-02:
#0  0x00409624 in lv_label_design (label=0x48fb90 <work_mem+828>, 
    mask=0x77eeeeb8, mode=<optimized out>) at lvgl/lv_objx/lv_label.7
667     lvgl/lv_objx/lv_label.c: No such file or directory.
(gdb) bt
#0  0x00409624 in lv_label_design (label=0x48fb90 <work_mem+828>, 
    mask=0x77eeeeb8, mode=<optimized out>) at lvgl/lv_objx/lv_label.7
#1  0x00407250 in lv_refr_obj (obj=0x48fb90 <work_mem+828>, 
    mask_ori_p=mask_ori_p@entry=0x77eeef30) at lvgl/lv_core/lv_refr.5
#2  0x004072f4 in lv_refr_obj (mask_ori_p=<optimized out>, obj=<opti)
    at lvgl/lv_core/lv_refr.c:443
#3  lv_refr_obj_and_children (top_p=0x48f9d0 <work_mem+380>, 
    mask_p=mask_p@entry=0x77eeef30) at lvgl/lv_core/lv_refr.c:416
#4  0x00407b60 in lv_refr_area_part_vdb (area_p=0x48c480 <inv_buf>)
    at lvgl/lv_core/lv_refr.c:339
#5  lv_refr_area_with_vdb (area_p=0x48c480 <inv_buf>)
    at lvgl/lv_core/lv_refr.c:303
#6  lv_refr_areas () at lvgl/lv_core/lv_refr.c:250
#7  lv_refr_task (param=<optimized out>) at lvgl/lv_core/lv_refr.c:17
#8  0x0040c484 in lv_task_exec (lv_task_p=<optimized out>)
    at lvgl/lv_misc/lv_task.c:271
#9  lv_task_handler () at lvgl/lv_misc/lv_task.c:101
#10 0x00402e50 in gui_tick () at src/gui.c:148
#11 0x00401e10 in phone_time () at src/soc.c:393

function : gui_tick is 
```
void gui_tick() {
	lv_tick_inc(1);
	lv_task_handler();
}
```
function : phone_time() is:
```
 while (1) {
    		gui_tick();
    		usleep(5000);
    	}
```

kisvegabor, 2018-04-02:
Thank you! I will check your code in more detail tomorrow.

kisvegabor, 2018-04-03:
I checked you `gui_init()` function which looks fine. When I tested you GUI I needed to modify some things because I don't have the images and fonts. Finally the `gui_init()` ran for me. 

1. Have call any of your `.._set_...` functions when you experience the issue?
2. Can you reduce the scope by creating less objects in `gui_init()`?

xiaoxinda, 2018-04-03:
because i need some icon keep changes: wifi icon,gsm icon and times
any one icon is changed and frequently.
when i experience the issue,i setting time fonts to lcd.
It will run for nearly 15 hours and it will have a segfault.

xiaoxinda, 2018-04-03:
If I can't find the problem, the only way I can temporarily solve it is to run it for a certain period of time, kill the process, and start over.

kisvegabor, 2018-04-03:
> It will run for nearly 15 hours and it will have a segfault.

I sounds like a memory leak. Please run this periodically (every 1..2 sec):

```c

    lv_mem_monitor_t mon;
    lv_mem_monitor(&mon);
    printf("used: %6d (%3d %%), frag: %3d %%, big free: %6d\n",
                                              (int)mon.total_size - mon.free_size,
    		                              mon.used_pct,
					      mon.frag_pct,										 
                                             (int)mon.free_biggest_size);
``` 

It will print the memory usage of LittlevGL's internal memory pool.

xiaoxinda, 2018-04-03:
already running
Print out this information
```
used:   1636 (  5 %), frag:   1 %, big free:  31104
used:   1636 (  5 %), frag:   1 %, big free:  31104
used:   1636 (  5 %), frag:   1 %, big free:  31104
used:   1636 (  5 %), frag:   1 %, big free:  31104
used:   1636 (  5 %), frag:   1 %, big free:  31104
used:   1636 (  5 %), frag:   1 %, big free:  31104
used:   1636 (  5 %), frag:   1 %, big free:  31104
used:   1636 (  5 %), frag:   1 %, big free:  31104
used:   1636 (  5 %), frag:   1 %, big free:  31104
used:   1636 (  5 %), frag:   1 %, big free:  31104
used:   1636 (  5 %), frag:   1 %, big free:  31104
used:   1636 (  5 %), frag:   1 %, big free:  31104
used:   1636 (  5 %), frag:   1 %, big free:  31104
```
wait for a segmentation error before watching

kisvegabor, 2018-04-03:
Great!
The fragmentation is not increasing?

xiaoxinda, 2018-04-03:
maybe not
Still it is:
```
used:   1636 (  5 %), frag:   1 %, big free:  31104
used:   1636 (  5 %), frag:   1 %, big free:  31104
used:   1636 (  5 %), frag:   1 %, big free:  31104
used:   1636 (  5 %), frag:   1 %, big free:  31104
used:   1636 (  5 %), frag:   1 %, big free:  31104
used:   1636 (  5 %), frag:   1 %, big free:  31104
used:   1636 (  5 %), frag:   1 %, big free:  31104
used:   1636 (  5 %), frag:   1 %, big free:  31104
used:   1636 (  5 %), frag:   1 %, big free:  31104
```

kisvegabor, 2018-04-03:
It seems no memory operations are happening.

Do you call e.g. `wifi_state` from an other thread?

xiaoxinda, 2018-04-03:
yes it is.
```
#include <stdio.h>
#include <sys/socket.h>
#include <arpa/inet.h>
#include <string.h>
#include <errno.h>
#include <stdlib.h>
#include <unistd.h>
#include <signal.h>
#include <time.h>
#include <limits.h> 
#include <pthread.h>
#include <sys/types.h>
#include <sys/stat.h> 
#include <fcntl.h>  
#include <termios.h>  
#include "../include/PhoneDateExtract.h"
#include "../include/gui.h"
#include "../include/uart.h"
int listener_d,time_type=0,is_out = 0;
time_t iit;
time_t t;  
char str_time[20];
time_t t; 
struct tm *lt;
#define USB_DRIVER0 "/dev/ttyUSB3"
#define USB_DRIVER1 "/dev/ttyUSB2"
#define USB_DRIVER2 "/dev/ttyS1"

// 错误处理函数
void error(char *msg) {
    fprintf(stderr, "Error: %s  %s", msg, strerror(errno));
    exit(1);
}

// 开启socket
int open_listener_socket() {
    int s = socket(PF_INET, SOCK_STREAM, 0);
    if (s == -1) {
        error("Can't open socket");
    }
    return s;
}

// 绑定端口
void bind_to_port(int socket, int port) {
    struct sockaddr_in name;
    name.sin_family = PF_INET;
    name.sin_port = (in_port_t)htons(port);
    name.sin_addr.s_addr = htonl(INADDR_ANY);
    int reuse = 1;
    if (setsockopt(socket, SOL_SOCKET, SO_REUSEADDR, (char *)&reuse, sizeof(int)) == -1) {
        error("Can't set the reuse option on the socket");
    }
    int c = bind(socket, (struct sockaddr*)&name, sizeof(name));
    if (c == -1) {
        error("Can't bind to socket");
    }
}

// 向客户端发消息
int say(int socket, char *s) {
    int result = (int)send(socket, s, strlen(s), 0);
    if (result == -1) {
        fprintf(stderr, "%s: %s \n","和客户端通信发生错误",strerror(errno));
    }
    return result;
}

[... 404 lines cut ...]
                    }else if(strstr(p1,"mynum") != NULL){
                        gui_set_phonenumber(p2);
                    }
                    free(p1);
                    p1 = NULL;
                    free(p2);
                    p2 = NULL;

                    say(connect_d,"is ok");
                }else{
                    say(connect_d,"input is  error");
                }
                if(NULL != buff){
                    free(buff);
                    buff = NULL;
                }
                close(connect_d);
        
    }
    

    return 0;
}

```

kisvegabor, 2018-04-03:
I see. I possible issue might be that the task scheduler interrupts LittelvGL in a wrong time and you modify something. When LittlevGL continues to run something might be broken.

I can be solved easily if you periodically call the WiFi state from a a LittlevGL instead of call LittlevGL functions externally.

You can create a simple LittlevGL task and call the states there:
```c
lv_task_create(my_state_updater, 100, LV_TASK_PRIO_MID, NULL);

 
void my_state_updater(void * p)
{
    int ws = wifi_state_from_driver();
    
    wifi_state(ws);
}
```

xiaoxinda, 2018-04-03:
Ok, I give it a try, thanks

xiaoxinda, 2018-05-17:
According to what you said, this problem has been solved.

kisvegabor, 2018-05-17:
Great! Thank you very much for the feedback!

closed 2018-05-17 by kisvegabor