lvgl/lvgl #182: Image as a button

Masi2017, 2018-04-02:
I was wondering if it is possible to have an image as an button or how can I add an image as a symbol and then assign that to the button text?

xiaoxinda, 2018-04-03:
https://littlevgl.com/image-to-c-array
This page is a img to c array tools
if your button is a transparent background image, through it to convert to alpha channel

kisvegabor, 2018-04-03:
It's a planned feature in v5.2 to have an object type similar to a button but show different images in each state.

Masi2017, 2018-04-03:
@kisvegabor Sounds good , it can be a really useful feature. Thank you

Masi2017, 2018-04-03:
@xiaoxinda Sorry I didn't get that, I am gonna use lvgl button object but add an image to the button, but since the image background might be different from the button ( button has two different background in press or release state) it doesn't look good.

kisvegabor, 2018-09-06:
Hi,

I added the `lv_imgbtn` object to `dev-5.2` branch. You can set different images in each state with `lv_imgbtn_set_src()`. The add actions use `lv_imgbtn_set_action()`

If you update `lvgl` to `dev-5.2` you also should update `lv_examples` to `dev-5.2` and convert your images again with the [new image converter tool](http://www.test.littlevgl.com/image-to-c-array) .

For clearence I close this issue, but of tested it and something not works feel free to reopen it and comment here.

closed 2018-09-06 by kisvegabor