lvgl/lvgl #183: Auto-fit feature in container object

BenQoo, 2018-04-03:
If enable auto-fit feature through `lv_cont_set_fit`, there are some issue about container's children:
* The position of children will be incorrect. (through `lv_obj_set_pos` or `lv_cont_set_layout`, `lv_obj_align`)
* The value of children' height will be incorrect. (through `lv_obj_get_height`)

I only tested the vertical auto-fit feature. I guess there is the same issue on horizontal fit portion.

Example (enable vertical auto-fit)
```C
lv_coord_t cont_y = 0;

lv_style_plain_color.body.border.width = 1;
lv_style_plain_color.body.border.color = LV_COLOR_RED;
lv_style_plain_color.body.opa = LV_OPA_50;

for (uint8_t n = 0; n < 4; n++) {
    lv_obj_t * container = lv_cont_create(lv_scr_act(), NULL);
    lv_cont_set_style(container, &lv_style_plain_color);
    lv_obj_set_size(container, 200, 45);
    lv_obj_set_pos(container, 0, cont_y);

    // enable vertical auto-fit
    lv_cont_set_fit(container, false, true);

    cont_y += 80;

    /* creeate button at top-right side */
    lv_obj_t * btn = lv_btn_create(container, NULL);
    lv_obj_align(btn, container, LV_ALIGN_IN_TOP_RIGHT, 0, 0);      // abnormal

    /* create label at top-left side */
    lv_obj_t * label_title = lv_label_create(container, NULL);
    lv_label_set_text(label_title, "Title");
    lv_obj_set_pos(label_title, 0, 0);      // abnormal

    lv_obj_t * label_descr = lv_label_create(container, NULL);
    lv_label_set_text(label_descr, "description\nABCDE");
    lv_obj_align(label_descr, label_title, LV_ALIGN_OUT_BOTTOM_LEFT, 0, 0);
}
```
LCD Size: 480x272 

Without vertical auto-fit
[image]

With vertical auto-fit
[image]

Besides, about the following code in `lv_cont_set_fit`.
The third parameter is assigned to `cont`, it will be converted to `lv_area_t *` type at signal function. Is it the expected behavior?
<https://github.com/littlevgl/lvgl/blob/9b2de5f215bba290a811a06dc9df084ba7936c23/lv_objx/lv_cont.c#L138>

Thank you!

kisvegabor, 2018-04-03:
Hi,

Thank you for the great code example and image. 

You experience it because you have `body.padding.ver != 0`. When you set something to y == 0 coordinate the library tries to keep `body.padding.ver` padding above the object. So modifies the parent's y0 coordinate with `body.padding.ver`. If you follow the positioning in your code step-by-step (with this in mind) the behavior become logical.

To avoid it you can use: `lv_style_plain_color.body.padding.ver = 0;`

So arbitrary positioning with auto fit will cause this effect. Positioning siblings to each other with `lv_obj_align` should work better.

Try this:
```c
    lv_obj_t * cont = lv_cont_create(lv_scr_act(), NULL);
    lv_obj_set_pos(cont, 100, 100);
    lv_cont_set_fit(cont, true, true);

    lv_obj_t * o1 = lv_obj_create(cont, NULL);
    lv_obj_set_drag(o1, true);

    lv_obj_t * o2 = lv_obj_create(cont, NULL);
    lv_obj_set_drag(o2, true);
    lv_obj_set_pos(o2, 50, 50);
```

Drag the blue rectangle to see how the parent size changes.

I fixed the line you mentioned in `beta` and a few other things too. Please use the `beta` branch to test the updates.

BenQoo, 2018-04-10:
Hi,
Thanks for your reply, I got it.
The `lv_cont` will change its position and size dynamically if enable the auto-fit feature.
Please let me close those issues, thank you!

closed 2018-04-10 by BenQoo