lvgl/lvgl #186: Text obj delete in keyboard action

Masi2017, 2018-04-04:
Assume that I have created an screen with some buttons there once user click on one of the button a keyboard object appears , after type the desired text and press ok button , the main screen should appears , In OK action I delete the keyboard obj but still text area has remained and I don't have access to that in OK action callback function, for now I created a global variable for text area so I am able to delete it in keyboard callback function but I don't know what is the best way or there is other solutions , I hope I have had some helps here. 
I tried to get text object from the keyboard and then delete the text but it didn't work and system crashed :  lv_obj_del(lv_kb_get_ta(kb))
Thanks in advance

kisvegabor, 2018-04-05:
Once you assigned a Text area to Keyboard with `lv_kb_set_ta()` you can get that Text area in the action with `lv_kb_get_ta()`.

Masi2017, 2018-04-05:
I did that but then system crashed and I got some error in adding object to lv_group. Basically when in a function I delete previous objects and then create another one and add object to group I will get errors and system crashed. I don't know maybe because the object delete still in progress and once I create a new abject and add to the group previews one still hasn't been deleted.

kisvegabor, 2018-04-07:
Pleas show a simple code example to reproduce the issue.

Masi2017, 2018-04-13:
I think problem is because in lv_app_kb_action , text is de-assigned from the keyboard and that's why it's not accessible in ok_action or close_action. if lv_kb_get_ta is called in  either ok_action or close_action callback function , system is crashed.

kisvegabor, 2018-04-13:
You are right! I fixed it in `beta`. Is it working now?

Masi2017, 2018-04-13:
For now I defined a global variable as an ta object and I get the text directly from the ta object. also I can edit lv_app_kb_action and remove de-assigned part if there is ok or close action.
I am good for now , thank you!

kisvegabor, 2018-04-15:
Okay, than you for the feedback.
Then I close this issue.

closed 2018-04-15 by kisvegabor