lvgl/lvgl #188: [Suggestion] Label scroll/roll improvements

upbeat27, 2018-04-04:
This is in reference to both `LV_LABEL_LONG_SCROLL` and `LV_LABEL_LONG_ROLL`.

I noticed the label's x location was not consistent between when a label was and was not scrolling.

This is due to:
```
anim.start = lv_font_get_width(font, ' ');
```
in the animation used for the scrolling/rolling. I assume the intent of this was to add a bit of padding so that when it started scrolling, the first character wasn't immediately disappearing from view.
This makes some sense, but I think it is the wrong way to achieve this.

In my opinion it works much better to do:
```
anim.start = 0;
anim.act_time = -anim.playback_pause;
```
for both the scrolling and rolling modes. The addition of the delay prevents the rolling from starting immediately, which is jarring when the label text changes and it is immediately rolling.

I would also suggest that the pause time (used for `playback_pause`, `repeat_pause`, and `act_time` in my example) be settable just like the animation speed. As is, it is too fast for my liking, and I can imagine others might have their own preference as well. At the very least maybe `ANIM_WAIT_CHAR_COUNT` can be broken out into `lv_conf.h` similar to `LV_LABEL_SCROLL_SPEED` (perhaps with a better name though). It wouldn't be settable at run time, but at least it would be changeable without modifying source.

Thoughts?

kisvegabor, 2018-04-05:
Hi,

You are right the intent was to add a bit of padding but not only to avoid immediate disappearing. It simply looks better (in my opinion) if the label not starts immediately on the first pixel. Because of the border of the parent or an adjacent object it might look ugly. What do you think?

So I think it's good to add a bit of padding but it's a good idea to make thinks configurable. The padding values can came from the style's pading values. (however they are also used when  lv_label_set_body_draw is used.) Or we can make possible to set a character which width will be the padding (now `' '`). You can set '\0' too to create zero padding.

upbeat27, 2018-04-05:
My issue with the padding is that when the label is not long enough to scroll, its position is different than when it is long enough to scroll. This makes the look inconsistent.

For example, let's say you have some labels in a column, all left justified:
```
|  Label 1          |
|  Label 2          |
|  Label 3          |
```
Now if one is long enough to roll, it isn't left justified anymore, because of the one space width added:
```
|  Label 1          |
|  Label 2          |
|   Label 3 is longe|r than the others
```
Which is not something you would expect to happen.

If the padding is made configureable, that would help!

What do you think about the delay before animation starts? Right now it starts immediately which doesn't look too great when the label text changes, or entering a new menu. Making it configureable like `LV_LABEL_SCROLL_SPEED` would also make it more flexible (the pause/wait/delay; they can be the same value probably).

kisvegabor, 2018-04-07:
Now I got it and agree with you. Both the updates seem useful. 

Would you like to also implement these features in dev-5.2 branch?

kisvegabor, 2018-07-25:
We had a discussion about reworking style system to be more CSS-like: https://github.com/littlevgl/lvgl/issues/302

Once it will be added in v6.0 the margin attribute should be used for this purpose.

kisvegabor, 2018-12-18:
Hi,

I was thinking about this issue again and realized that `LV_LABEL_LONG_SCROLL` (moving the object) is pointless because `LV_LABEL_LONG_ROLL` is more flexible. By setting the size and position of the label the required padding can be achieved using `.._ROLL`

So I removed the extra `' '` as you suggested. Probably `LV_LABEL_LONG_SCROLL` will be removed in `v6.0` because `LV_LABEL_LONG_ROLL` is better.

Thank you for pointing this out.

closed 2018-12-18 by kisvegabor