lvgl/lvgl #189: Have somebody ported LittlevGL to Stm32duino?

NickNifontov, 2018-04-05:
Have somebody ported LittlevGL to Stm32duino?

http://wiki.stm32duino.com/index.php?title=Main_Page
http://www.stm32duino.com/

kisvegabor, 2018-04-05:
Hi,

Once I created an Arduino library from LittlevGL. You can find it here: https://github.com/littlevgl/lv_boards/tree/master/arduino_library

I didn't know Stm32duino but I saw it has STM32F103 board has better support. Unfortunately it has only a few Flash. A simple LittlevGL build is about 50kB.

NickNifontov, 2018-04-05:
STM32F103CBT8 - 128kb flash.
STM32F103RC - 512 kb flash

seem size is ok

kisvegabor, 2018-04-07:
Then you can try the mentioned Arduino example. You just need to update it becase it is still from v5.0

kisvegabor, 2018-05-17:
I updated the Aurdino example to v5.1. Could you try it?

NickNifontov, 2018-05-17:
ok. i'll try

rbultman, 2018-07-11:
I've been messing around with porting to the [STM32F446RE Nucleo board](https://www.st.com/en/evaluation-tools/nucleo-f446re.html) and the [Adafruit TFT touch LCD](https://www.adafruit.com/product/2090).  It is basically working.  The demo app works fine but the keyboard tutorial locks up with the popup is dismissed.

kisvegabor, 2018-07-11:
"locks up" means when popup is closed the GUI freezes? Or the whole MCU freezes?

rbultman, 2018-07-11:
The popup doesn't actually close.  The whole MCU freezes.  I haven't tracked down where, but it seems to be related to the group.  If you don't add the mbox to the group, the mbox dismisses and the app continues to run just fine.

kisvegabor, 2018-07-11:
Please, update lvgl to the `beta` branch. There were some bugfixes related to groups. Hope it's already fixed.

rbultman, 2018-07-11:
Will do.  BTW, pretty cool library.

rbultman, 2018-07-11:
@kisvegabor I'm still seeing the same behavior.  I may try a straight gnu toolchain to see if I can get it working with this setup.  Alternatively, I just got an STM32F429I-DISC1 board to try.  Hopefully your example with the older version of this board will work just fine.

kisvegabor, 2018-07-13:
I checked keyboard tutorial on STM32F429 Discovery and it worked. I used the demo project from the lv_board repo.
However, I made the button to be the first object because it wasn't visible on the 240x320 portrait-oriented screen.

rbultman, 2018-07-13:
@kisvegabor I did the same thing and also confirmed.  I'm trying to get the library working on the Nucleo 446RE board but using the ST MX Cube tools instead of Arduino.  I'm still using the Adafruit LCD I referenced earlier.  I'm just trying to see if it works with different hardware but the same toolchain as used for the STM32F429.  If it does, then I can try to figure out why the problem happens with the Arduino toolchain.

epikao, 2018-07-18:
hello, as mentioned by kisvegabor above, I have a similar problem with the littlevgl library and arduino (teensy 3.2).
The problem is that, as soon as I activate the "lv_tick_inc(5);" with a Timer or in the loop the program freezes, crashs. Nothing more is working, no loop nothing. During compiling there is no error or something shown.
I can remember to have a similar problem if I setup a local array which use too much RAM.... 

But anyway where can I  find and check what the "lv_tick_inc(5);" exactly do? Thank you

epikao, 2018-07-18:
I just tried, but it does not help. I think is not a memory problem, because with standard value it shows me still 35644 bytes free for the dynamic memory. So that must be enough!

With your setup in conf.h compiler show me 47676 free bytes for dynamic memory.

kisvegabor, 2018-07-18:
I see. Thank for the testing. 

In the `dev-5.2` branch I add the `lv_log` module but not applied it in lvgl so far. I will do it soon then we will see which task cause the crash.

epikao, 2018-07-19:
> In the dev-5.2 branch I add the lv_log module but not applied it in lvgl so far. I will do it soon then we will see which task cause the crash.

oh, that would be very helpful - hope you will find soon time for that :-)

BTW: I also remove all Serial (Debug) Interface output because it's well known that it can also cause crash. But that not solved the problem. So program is still freezing/crashing.

kisvegabor, 2018-07-25:
Hi,

I just added some logs to the code. In order to use it:
1. Upgrade `lvgl` from `dev-5.2` branch 
2. Enable logging in `lv_cond.h`. See `lv_conf_templ.h`: https://github.com/littlevgl/lvgl/blob/dev-5.2/lv_conf_templ.h#L92
3. Set `LV_LOG_LEVEL` to `LV_LOG_LEVEL_TRACE`
4. Register a function to print the log to the console: `lv_log_register_print(my_func)`. Example: https://github.com/littlevgl/lvgl/blob/dev-5.2/lv_misc/lv_log.c#L57
5. Compile, run, see what happens :)

epikao, 2018-07-26:
Hi, thank you.
in the lv_conf.h (arduino librarie) the Log setting code does not exist:
```
/*Log settings*/
#define USE_LV_LOG		1	/*Enable/disable the log module*/
#if USE_LV_LOG
/* How important log should be added:
 * LV_LOG_LEVEL_TRACE		A lot of logs to give detailed information
 * LV_LOG_LEVEL_INFO		Log important events
 * LV_LOG_LEVEL_WARN		Log if something unwanted happened but didn't caused problem
 * LV_LOG_LEVEL_ERROR		Only critical issue, when the system may fail
 */
#define LV_LOG_LEVEL	LV_LOG_LEVEL_INFO
/* 1: Print the log with 'printf'; 0: user need to register a callback*/ 
```

Should I simply add that code in the conf-file?

thanks

kisvegabor, 2018-07-26:
Yes, you should just update it.

kisvegabor, 2018-07-26:
BTW, I found a serious issue when initialized the memory manager. I wonder how it could remain uncovered so long. Basically, the library overwrote some data on arbitrary places. So maybe `dev-5.2` simply will work :)

epikao, 2018-07-26:
> Upgrade lvgl from dev-5.2 branch:

I simply replaced the old files in the libraries with the files from dev-5.2 branch.
Now I have follow error during compiling see below:
[image]

kisvegabor, 2018-07-26:
Do you see `lv_area_intersect()` in `lv_misc/lv_area.c` and `lv_misc/lv_area.h`?

epikao, 2018-07-26:
> Do you see lv_area_intersect() in lv_misc/lv_area.c and lv_misc/lv_area.h?

yes that function I have in both files, what do you mean with that?

kisvegabor, 2018-07-26:
Have you deleted the old lvgl and added the new, or just merged the two lvgl folder?

From the error message it seems that some files from the previous version is still there.

kisvegabor, 2018-07-26:
I tried to compile it with my teensy project and it worked for me.

I will exemine it deeper.

kisvegabor, 2018-07-27:
Hi,

I tested it again. A half part was my bad the other part was strange:
1. Please update lvgl again. I pushed some updates
2. Open `lv_misc/lv_area.c` modify `lv_area_get_size` to `lv_area_get_size2`. Try to compile (will fail), rename `lv_area_get_size2` back to the original. Somehow Arduino didn't realize the new content of this file but this trick solved it for me.

kisvegabor, 2018-07-27:
Look the difference. Your code snippet:
```c
/*Log settings*/
#define USE_LV_LOG		1	/*Enable/disable the log module*/
#if USE_LV_LOG
/* How important log should be added: */
LV_LOG_LEVEL_TRACE		//A lot of logs to give detailed information
LV_LOG_LEVEL_INFO		//Log important events
LV_LOG_LEVEL_WARN		//Log if something unwanted happened but didn't caused problem
LV_LOG_LEVEL_ERROR		//Only critical issue, when the system may fail
#define LV_LOG_LEVEL	LV_LOG_LEVEL_TRACE
/* 1: Print the log with 'printf'; 0: user need to register a callback*/

#define LV_LOG_PRINTF	0
#endif  /*USE_LV_LOG*/
```

My snippet is(`LV_LOG_LEVEL_...` lines are in comment): 
```c
/*Log settings*/
#define USE_LV_LOG		1	/*Enable/disable the log module*/
#if USE_LV_LOG
/* How important log should be added:
 * LV_LOG_LEVEL_TRACE		A lot of logs to give detailed information
 * LV_LOG_LEVEL_INFO		Log important events
 * LV_LOG_LEVEL_WARN		Log if something unwanted happened but didn't caused problem
 * LV_LOG_LEVEL_ERROR		Only critical issue, when the system may fail
 */
#define LV_LOG_LEVEL	LV_LOG_LEVEL_TRACE
/* 1: Print the log with 'printf'; 0: user need to register a callback*/

#define LV_LOG_PRINTF	0
#endif  /*USE_LV_LOG*/
```

epikao, 2018-07-27:
ok x-) , sorry for the circumstances I was thinking that I must enable something in that log ^^, now it compiles without error.

AND now it works without freezes/crash, but it looks that the "my_disp_flush" is still never called... respectively there is also nothing on the display.

epikao, 2018-07-27:
ok x-) , sorry for the circumstances I was thinking that I must enable something in that log ^^, now it compiles without error.

AND very great, now it works without freezes/crash 👍 , and the flush function is also called see below. 
The only problem is that I still see nothing on the display... but that will be another problem.

Anyway thank you very much for that support, great job.

[image]

epikao, 2018-07-27:
as I need to call the Lv_task_handler(); every 5ms, I think is better to call it in a timer instead in the main-loop?
Can I call it in the same timer as the tick function is called?

kisvegabor, 2018-07-27:
**lv_task_handler**
It should NOT be called on a higher priority than where you create the objects. For example. if create objects in the `setup` or `loop` then `lv_task_handler` can be called during creating the objects and cause errors. It'"s safer to call it on the lowest priority which is the `loop` in your case.

**color array**
You can think about it as an array of 16bit integers. You can cast is as well. 
```c
uint8_t * color_array_u8 = (uint8_t*) color_array;
SPI.transfer(color_array_u8[c]);
```
But there is an interesting thing:
1. Let say there is a red color in the color array: 0xF800 (RGB565)
2. As you MCU is Little Endian the 0xF800 value will be stored in the memory as 0x00 (lower address), 0xF8 (higher address)
3. So when you send the color array form byte-to-byte the LSB of the color will be sent first
4. If your display controller waits the MSB first then color will be wrong.
5. To solve it:
   a) Configure the display controller to wait the MSB first (if it is supported)
   b) Send 16bit data on SPI (if supported)
   c) Swap the bytes of all colors before sending them
   d) Send the `c+1` byte forst via SPI: 
       `SPI.transfer(color_array_u8[c + 1]);  SPI.transfer(color_array_u8[c]); `

But don't deal with it now. If it works you should see something just with wrong colors.

epikao, 2018-07-27:
`
> Swap the bytes of all colors before sending them
`

I already have a function which swap the bytes, I need it also before - so I think that problem I already have solved. So now I will try the color array issue and will feedback you later. thanks

epikao, 2018-07-27:
that's what I get at the moment with follow SPI transfer code:

```
  uint8_t * color_array_u8 = (uint8_t*) color_array;
  SPI.transfer(color_array_u8, 0, sizeof(color_array_u8));
```

[image]

kisvegabor, 2018-07-27:
It's not correct: `SPI.transfer(color_array_u8, 0, sizeof(color_array_u8));`
`sizeof(color_array_u8)` == 4 because it's a pointer

`SPI.transfer(color_array_u8, 0, (x2 - x1 + 1) * ((y2 - y1 + 1)));`

kisvegabor, 2018-07-27:
Pointers: https://www.tutorialspoint.com/cprogramming/c_pointers.htm

Have you selected the area on the display before sending data there?

epikao, 2018-07-27:
> Have you selected the area on the display before sending data there?

what do you mean with that? In the init the area is define for the whole area 240x320.
And follow functions which I have create myself works properly:
```
void colorxy(uint16_t x1, uint16_t x2, uint16_t y1, uint16_t y2, uint8_t r, uint8_t g, uint8_t b)
{
  uint16_t color16 = ((r>>3)<<11)+((g>>2)<<5)+(b>>3);
  color16 = ((color16<<8)&0xff00)|((color16>>8)&0x00ff); //swap bytes

  uint16_t rgb[7500]; //5846, 6952
  for(uint16_t c=0; c<7500; c++)
  {rgb[c]=color16;}
  
  writecommand(0x2A); //x pixel-position 
  writedata(x1>>8); //start-position
  writedata(x1); //start-position
  writedata(x2>>8); //end-position
  writedata(x2); //end-position

  writecommand(0x2B); //y pixel-positon 
  writedata(y1>>8); //start-position
  writedata(y1); //start-position
  writedata(y2>>8); //end-position
  writedata(y2); //end-position

  writecommand(0x2C); //send RAM Data
  digitalWrite(CS0, 0);
  digitalWrite(DC, 1); //Command = 0, Data = 1
  SPI.beginTransaction(settingsA);
  SPI.transfer(rgb, 0, sizeof(rgb));
  SPI.endTransaction();
  //SPI.transfer(color);
  digitalWrite(CS0, 1);  
}
```

kisvegabor, 2018-07-27:
```c
writecommand(0x2A); //x pixel-position 
  writedata(x1>>8); //start-position
  writedata(x1); //start-position
  writedata(x2>>8); //end-position
  writedata(x2); //end-position

  writecommand(0x2B); //y pixel-positon 
  writedata(y1>>8); //start-position
  writedata(y1); //start-position
  writedata(y2>>8); //end-position
  writedata(y2); //end-position
```

You need to apply this `disp_flush` too:

Please use  ` ```c   ... ``` ` for code blocks to add syntax highlight.

epikao, 2018-07-27:
> You need to apply the above in the disp_flush too!

that code was always in the flush function... hmm

kisvegabor, 2018-07-27:
Then please test the disp_flsuh function like this:

```c
	
	lv_color_t c_test[1000];
	uint16_t i;
	
	/*RED top left corner*/
	for(i = 0; i < 1000; i++) c_test[i] = LV_COLOR_RED;
	
	disp_flush(0,0, 20, 40, c_test);
	
	/*BLUE middle*/
	for(i = 0; i < 1000; i++) c_test[i] = LV_COLOR_BLUE;
	
	disp_flush(100, 100, 130, 130, c_test);
	
	
	/*Green vertical line*/
	for(i = 0; i < 1000; i++) c_test[i] = LV_COLOR_GREEN;
	
	disp_flush(80, 30, 80, 130, c_test);
	
	/*YELLOW horizontal line*/
	for(i = 0; i < 1000; i++) c_test[i] = LV_COLOR_YELLOW;
	
	disp_flush(10, 30, 100, 30, c_test);

```

epikao, 2018-07-27:
first I clear the RAM with 0xFF bytes (make screen white) and than I call your test function and the result is follow:

[image]

kisvegabor, 2018-07-27:
It seems the `disp_flush` function is not working well. Copy it here please.

epikao, 2018-07-27:
```c
void my_disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t *color_array)
{

  /*TODO copy 'color_array' to the specifed coordinats of your disaply.*/
  
  writecommand(0x2A); //x pixel-position anfahren
  writedata(x1>>8); //start-position
  writedata(x1); //start-position
  writedata(x2>>8); //end-position
  writedata(x2); //end-position

  writecommand(0x2B); //y pixel-positon anfahren
  writedata(y1>>8); //start-position
  writedata(y1); //start-position
  writedata(y2>>8); //end-position
  writedata(y2); //end-position

  writecommand(0x2C); //RAM Daten senden
  digitalWrite(CS0, 0);
  digitalWrite(DC, 1); //Command = 0, Data = 1

  //SPI.transfer(color_array, 0, sizeof(color_array));
  uint8_t * color_array_u8 = (uint8_t*) color_array;
  SPI.transfer(color_array_u8, 0, (x2 - x1 + 1) * ((y2 - y1 + 1)));
  
  digitalWrite(CS0, 1);
  /*Tell the flushing is ready*/
  lv_flush_ready();
}
```

kisvegabor, 2018-07-27:
`digitalWrite(CS0, 0);` should be in the beginning, isn't it?

epikao, 2018-07-27:
> digitalWrite(CS0, 0); should be in the beginning, isn't it?
is not needed, because in the command / data function already implemented.
In my function colorxy mentioned previous post I do it exactly the same, and that works properly.

```c
void writecommand(uint8_t a)
{
  digitalWrite(CS0, 0);
  digitalWrite(DC, 0); //Command = 0, Data = 1
  //SPI.begin();
  //SPI.beginTransaction(settingsA);
  SPI.transfer(a);
  //SPI.endTransaction();
  digitalWrite(CS0, 1);
}

void writedata(uint8_t a)
{
  digitalWrite(CS0, 0);
  digitalWrite(DC, 1); //Command = 0, Data = 1
  //SPI.begin();
  //SPI.beginTransaction(settingsA);
  SPI.transfer(a);
  //SPI.endTransaction();
  digitalWrite(CS0, 1);
}
```

kisvegabor, 2018-07-27:
I see.
Is it possible that `SPI.transfer` still sends data in the background when `disp_flush` exits? If yes it is possible that the last transfer is not finished when the new starts. The most simple solution to add a delay to the beginning of  `disp_flush` (300 ms for test) to be sure the last flush is ready.

epikao, 2018-07-27:
ok thanks, I will try that maybe this evening or tomorrow

epikao, 2018-07-27:
without lv_task_handler() it looks as follow:

[image]

epikao, 2018-07-27:
do you have a function which is called simply every pixel and which only transfer a simple 16bit color value and not a array?

kisvegabor, 2018-07-27:
My suggestion was buggy. In disp_flush try `SPI.transfer(color_array_u8, 0, (x2 - x1 + 1) * ((y2 - y1 + 1)) * sizeof(lv_color_t);`

How does it look like now?

To see the label you should call `lv_task_handler`. But to test `disp_flush` you can comment it out.

epikao, 2018-07-27:
>To see the label you should call >lv_task_handler. But to test disp_flush you >can comment it out

I will test you corrections tomorrow, thanks.
Do you also have a function which is simply called every pixel? It will make it easier for me to implement display driver

kisvegabor, 2018-07-27:
> Do you also have a function which is simply called every pixel? It will make it easier for me to implement display driver

No, it isn't because it would be very slow.

epikao, 2018-07-28:
now it looks like that:
[image]

if activate the lv-task handler, the lines/square will be overwritten white and I can see some pixel from the "Hello Arduino" function call..:

[image]

epikao, 2018-07-28:
BTW: Regarding the wrong color issue, is there no way to simply swap your 16bit color array like my function below:
color16 = ((color16<<8)&0xff00)|((color16>>8)&0x00ff); //swap bytes

kisvegabor, 2018-07-29:
Hi,

Have you set `LV_HOR_RES` and `LV_VER_RES` in lv_conf.h?

You can swap the bytes in the color array directly.
```c
    uint16_t color_array_16 = (uint16_t *) color_p;
    uint32_t i;
    for (i = 0; i <  (x2 - x1 + 1) * (y2 - y1 + 1); i++) {
    	color_array_16[i] = ((color_array_16[i] <<8 ) & 0xff00) | 
    			    ((color_array_16[i] >>8 ) & 0x00ff); //swap bytes
    }
```

epikao, 2018-07-29:
>color array
>You can think about it as an array of 16bit >integers. You can cast is as well.

>uint8_t * color_array_u8 = (uint8_t*) >color_array; >SPI.transfer(color_array_u8[c]);

the SPI.transfer also support 16bit, so can I use above Code but with number 16 instead 8?

kisvegabor, 2018-07-29:
If it supports 16 bit data then simply 
```c
uint16_t * color_array_16 = (uint16_t *) color_p;
SPI.transfer(color_array_16, 0, (x2 - x1 + 1) * (y2 - y1 + 1));   //If sending 16 bit don't multiply with sizeof(lv_color_t)
```

What about "Have you set LV_HOR_RES and LV_VER_RES in lv_conf.h?"

epikao, 2018-07-31:
thanks, but multiply with sizeof(lv_color_t) is needed, otherwise it looks wrong, so now see with sizeof:

[image]

> What about "Have you set LV_HOR_RES and LV_VER_RES in lv_conf.h?"

see:
/* Horizontal and vertical resolution of the library.*/
#define LV_HOR_RES          (320)
#define LV_VER_RES          (240)
#define LV_DPI              100

seems correct?
So the only one problem now is that lv_task_handler does overwrite big part with white, and I only see some pixel from "Hello Arduino" or what it should be

kisvegabor, 2018-08-01:
Awesome! :)

Your working with Teensy, right?
As this issue is about Stm32duino a confirmation is still required for that board. 
@rbultman can you test the `dev-5.2` branch of LittlevGL on Stm32duino?

epikao, 2018-08-03:
yes Teensy 3.2 :-)

[1 more reply]

closed 2018-09-04 by kisvegabor