lvgl/lvgl #190: Recolor on symbol and array in text

Masi2017, 2018-04-05:
Hi All,
Is that possible to set recolor mode for symbol and  raw character array  in a label? I think this works only in text.

kisvegabor, 2018-04-07:
Hi,
I don't clearly understand what your goal is. Can you show a code example?

Masi2017, 2018-04-09:
According to the document : It is possible to enable re-color parts of the text. For example: "Write a #ff0000 red# word". This feature can be enabled individually for each label by lv_label_set_recolor() function.
I was wondering if label's text is an array it's not possible to recolor using this feature?

kisvegabor, 2018-04-10:
It worked for me:
```c
    const char buf[] = {"Write a #ff0000 red# word"};

    lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);
    lv_label_set_recolor(label, true);
    lv_label_set_array_text(label, buf, strlen(buf));
```
[image]

Masi2017, 2018-04-10:
Thanks, sounds good .

kisvegabor, 2018-04-10:
Yes, you have to "insert" the command part manually an you also have to call `lv_label_set_array_text()` to refresh the size of the label.

Masi2017, 2018-04-10:
How about symbols? there is no way to use recolor feature for them right?

kisvegabor, 2018-04-11:
It is also valid:
```c
    const char buf[] = {"Write a #ff0000 "SYMBOL_OK"# word"};

    lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);
    lv_label_set_recolor(label, true);
    lv_label_set_array_text(label, buf, strlen(buf));
```

[image]

Masi2017, 2018-04-11:
@kisvegabor Thanks it worked fine.

closed 2018-04-12 by kisvegabor