lvgl/lvgl #193: How to display dynamic variables label:question

margce, 2018-04-09:
Hello,

I recently started working with LittleVGL, but there is still something I'm not sure how to accomplish. By looking at the examples and documentation labels seem to be pretty much static. What I mean is what happens if I want to display a text based on an integer or float variable?

Say that I want to display an ADC input value, what would be the best way to display it? As far as I understand, I need to setup a label but this approach requires to "manually" update it's content by determining which is the active screen and then setting the new content of the label.

What's the recommended approach for this?

kisvegabor, 2018-04-09:
Hi,

Very good question!

Now there is not "variable binding" or some thing like that in the library. The best to achieve it is to create a LittlevGL task and update the label there:

```c
lv_task_create(label_refresher_task, 100, LV_TASK_PRIO_MID, NULL);

...
    

uint32_t adc_value;

void label_refresher_task(void * p)
{
    static uint32_t prev_value = 0;

    if(prev_value != adc_value) {

        if(lv_obj_get_screen(my_label) == lv_scr_act()) {
            char buf[32];
            sprintf("Voltage: %d", adc_value);
            lv_label_set_text(my_label, buf);
        }
        prev_value = adc_value;
    }
}

```

margce, 2018-04-10:
Thanks, that's the term I was looking for "variable binding". 

I'll implement the solution you mentioned, the downside I see with this is that I have to create one task that's actively checking which screen is active and updating the required label. It would be handy if this is something that the library can do on it's own. Do you think think this can be added as an enhancement or it would be too complex?

Thanks,
Martin

closed 2018-04-13 by margce

effeci, 2018-04-17:
Hello! I'm a kind of a beginner in this field (arduino background).
I got a demo and some tutorials running on my board (stm32f746g disco) and now trying to understand how to implement this.
Basically i need some numerical values to update on the screen and the ability to switch trought menus. Do you have any clue to point me at?

kisvegabor, 2018-04-18:
Hi,

Have you tried the example code above?

effeci, 2018-04-18:
Hi! I don't know how to place it and where

effeci, 2018-04-22:
Thank you... I'll test that and report back

flossandmeditate, 2019-08-05:
Just wanted to check in a year later - what is the latest on variable binding? This still the best method?

embeddedt, 2019-08-05:
@flossandmeditate Yes, there are no plans to support auto-binding of variables at present. However, we are planning to add an `lv_label_set_text_fmt` function which would behave like `printf` but would change the text on a label. So you could update a label simply by doing this:

```c
int my_val = <whatever>;
lv_label_set_text_fmt(label, "%d", my_val);
```

It would be a lot more concise rather than putting it in a buffer first.

kisvegabor, 2019-08-05:
Or another way is:
1. set an event function for the label 
2. In `LV_EVENT_REFRESH` set the text as you need by using the variables you need
3. Send an `LV_EVENT_REFRESH` manually with `lv_event_send(label, LV_EVENT_REFRESH, NULL);`

It's not a variable binding but could help to better structure your code.

baldhead69, 2019-09-30:
Hi,

Does the "dinamically label" functionality already exist ?

Thank's.

embeddedt, 2019-09-30:
`lv_label_set_text_fmt` is in `dev-6.1` and works without issues from my tests.

baldhead69, 2019-09-30:
Thank's.

baldhead69, 2019-09-30:
Hi,

When the program calls this function the program resets the microcontroller (ep32).

void label_refresher_task( )
{
    static int16_t prev_value = 0;

    if(prev_value != bt1Time) 
    {
        if( lv_obj_get_screen(my_label) == lv_scr_act() )
        {
            char buf[32];
            sprintf("%d", bt1Time);
            lv_label_set_text(my_label, buf);
        }
        prev_value = bt1Time;
    }

}

embeddedt, 2019-09-30:
You aren't using `sprintf` property. Look up its prototype - you need to pass `buf` as the first parameter.

----------------------
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

It's possible to sign in to the forum with your GitHub account... you don't need to make a separate account.

baldhead69, 2019-09-30:
hahahaha.

Sorry.

I had followed the example of kisvegabor above, and had not even noticed that the syntax of the sprintf function was wrong.

Thank's.

embeddedt, 2019-09-30:
I edited his post so that no one else will make the same mistake. :slightly_smiling_face: 

If you have compiler warnings enabled, it should warn you about code like that at compile time rather than getting a crash at runtime.

kisvegabor, 2019-10-01:
> I had followed the example of kisvegabor above, and had not even noticed that the syntax of the sprintf function was wrong.

Ups, sorry. :sweat_smile: 

> I edited his post so that no one else will make the same mistake.

Thank you!

mbrdar, 2021-01-08:
> Hi,
> 
> Very good question!
> 
> Now there is not "variable binding" or something like that in the library. The best way to achieve it to create a LittlevGL task and update the label there:
> 
> ```c
> lv_task_create(label_refresher_task, 100, LV_TASK_PRIO_MID, NULL);
> 
> ...
>     
> 
> uint32_t adc_value;
> 
> void label_refresher_task(void * p)
> {
>     static uint32_t prev_value = 0;
> 
>     if(prev_value != adc_value) {
> 
>         if(lv_obj_get_screen(my_label) == lv_scr_act()) {
>             char buf[32];
>             snprintf(buf, 32, "Voltage: %d", adc_value);
>             lv_label_set_text(my_label, buf);
>         }
>         prev_value = adc_value;
>     }
> }
> ```

Is this still preferred way to update labels?

embeddedt, 2021-01-08:
@mbrdar https://docs.lvgl.io/latest/en/html/widgets/label.html#set-text

> With `lv_label_set_text_fmt(label, "Value: %d", 15)` printf formatting can be used to set the text.

DennisJacobs1992, 2022-10-02:
> > I had followed the example of kisvegabor above, and had not even noticed that the syntax of the sprintf function was wrong.
> 
> Ups, sorry. 😅
> 
> > I edited his post so that no one else will make the same mistake.
> 
> Thank you!

DennisJacobs1992, 2022-10-02:
> Or another way is:
> 
> 1. set an event function for the label
> 2. In `LV_EVENT_REFRESH` set the text as you need by using the variables you need
> 3. Send an `LV_EVENT_REFRESH` manually with `lv_event_send(label, LV_EVENT_REFRESH, NULL);` when the label needs to be refreshed.
> 
> It's not a variable binding but could help to better structure your code.

Hey @kisvegabor,

I am trying to change a lable at runtime doesn't seem to work out. I tryed your suggested solution but I am not sure if I am doing it correctly. Can you check my code? I am using LVGL 8.2.0 in the arudino IDE and created libraries where a ui_Label3 excists. I removed most of my code except for the event part for clearity. My code is compiling but it doesn't seem to update the label.

code:
________________________________________________________________________________________
int chargePercentage = 0;

static void chargePercentage_cb(lv_event_t * e);
void chargePercentageCounter();

void setup()
{
  lv_obj_add_event_cb(ui_Label3, chargePercentage_cb, LV_EVENT_REFRESH, NULL);
}

void loop() {
  lv_event_send(ui_Label3, LV_EVENT_REFRESH, NULL);
  lv_timer_handler(); /* let the GUI do its work */
  delay(50);
}

static void chargePercentage_cb(lv_event_t * e)
{
  lv_label_set_text_fmt(ui_Label3, "%d",chargePercentage);
}

void chargePercentageCounter()
{
  if (chargePercentage == 100)
  {
    chargePercentage = 0;
  } 
  else
  {
    chargePercentage++;
  }
}
________________________________________________________________________________________

kisvegabor, 2022-10-02:
Are you sure the UI is actually running? E.g. if you create a button can you press it and does it changes its color?

Have you configured the LVGL's tick as described [here](https://docs.lvgl.io/master/get-started/platforms/arduino.html#configure-lvgl)?

> Set LV_TICK_CUSTOM 1

DennisJacobs1992, 2022-10-03:
> Are you sure the UI is actually running? E.g. if you create a button can you press it and does it changes its color?
> 
> Have you configured the LVGL's tick as described [here](https://docs.lvgl.io/master/get-started/platforms/arduino.html#configure-lvgl)?
> 
> > Set LV_TICK_CUSTOM 1

You made my day. I forgot to set Set LV_TICK_CUSTOM 1. Its working now. Thanks a lot! Next time I should read the manual better.

https://user-images.githubusercontent.com/34340041/193645655-e3c3bcfd-b3e2-452f-b341-6522b2b14da8.mp4

https://user-images.githubusercontent.com/34340041/193645677-0d054b07-87c5-4ae0-9497-99ff16ddef54.mp4

kisvegabor, 2022-10-03:
Great! :)

baldhead69, 2023-04-04:
I do like this:
```
    lv_style_t style_screen1;

    lv_obj_t* scr1 = lv_obj_create(NULL);

    lv_style_init(&style_screen1);
    lv_style_set_bg_color(&style_screen1, lv_color_white());
    lv_obj_add_style(scr1, &style_screen1, LV_PART_MAIN );  //turn the screen white

    lv_obj_t* label_scr1 = lv_label_create(scr1);     
    lv_obj_set_style_text_align(label_scr1, LV_TEXT_ALIGN_CENTER, 0);
    lv_obj_align(label_scr1, LV_ALIGN_CENTER, 0, -100);        
    lv_label_set_text_fmt(label_scr1, "SCR1: %p", scr1);
```

kisvegabor, 2023-04-05:
You can take a look at [lv_msg](https://docs.lvgl.io/master/others/msg.html).

Syrinx98, 2023-04-05:
does lv_label_set_text_fmt create memory problems like lv_label_set_text? because when i use lv_label_set_text in a loop, after some time the screen freezes and i think it is because of a memory leak, does anyone know the best method to periodically refresh a variable? like a clock? thank you in advance.
@kisvegabor

kisvegabor, 2023-04-05:
Neither `lv_label_set_text` nor `lv_label_set_text_fmt` should have memory leak issues. Are you calling on this function in the loop?

brentru, 2023-04-05:
@kisvegabor  I have a similar issue (repeated calls to `lv_label_set_text` cause a hang) but am working on isolating it. Do you prefer issues like this (with code) here or in the forums?

Syrinx98, 2023-04-06:
@kisvegabor i am calling this function in a loop, but i don't understand why it should behave like so, how can i effectively update a text from my code without any problems? for example an information that i receive from modbus or gsm module.
Do i have to use lv_msg as you said? Thank you in advance

kisvegabor, 2023-04-06:
> Do you prefer issues like this (with code) here or in the forums?

Please open an issue for it here.

> i am calling this function in a loop, but i don't understand why it should behave like so, how can i effectively update a text from my code without any problems? for example an information that i receive from modbus or gsm module.

`lv_label_set_text` should work without any issues. And I'm pretty sure that the problem is not in this functions as it's very commonly used. If it were wrong we had many bug reports because of it. 

Please open a separate issue to track this and I'll help to figure out the root of the problem.

uched41, 2024-11-02:
I am also seeing this issue. Here is a video

https://github.com/user-attachments/assets/4c4991cc-5005-4127-9a7d-8cd7e33eca93

uched41, 2024-11-02:
Here is the code segment that is responsible for this:

```
static uint32_t prev = 234;
static char buffer[32];
while(1){
     k_msleep(2000);
     prev  += 29;
     printk("prev: %d\n", prev);
     snprintf(buffer, sizeof(buffer), "%d", prev);
     lv_label_set_text_fmt(base_scr.start_text_label, "%d", prev);
     v_task_handler();
}
```

As you can see in the video, the entire display gets corrupted during some updates. 

I have also confirmed that this issue is not in the display HAL layer (QSPI). I randomly updated an arc on the display and it updates without any issues.

Do you have any ideas on what could be the issue with the usage of `lv_label_set_text` or `lv_label_set_text_fmt` ?

vwheeler63, 2024-11-02:
Hi, @uched41 .

How are the pixels being transferred to the display?  Is it by DMA?

vwheeler63, 2024-11-02:
> No DMA involved.

Wow, so if I am seeing that correctly, every pixel has been sent to the physical display before `k_msleep(2000);` has a chance to execute again, am I correct?

uched41, 2024-11-02:
Yes, that is my understanding

```while(1){
      k_msleep(2000);
      prev  += 29;
      printk("prev: %d\n", prev);
      snprintf(buffer, sizeof(buffer), "%d", prev);
      lv_label_set_text_fmt(base_scr.start_text_label, "%d", prev);
      lv_task_handler();
  }
```

I'm assuming everything gets handled when I call `lv_task_handler`. 

I also tried to setup a recurring timer callback to call `lv_task_handler` every 5 or 10ms, but this didn't solve the problem.