lvgl/lvgl #196: Not able to see focused item in List

Masi2017, 2018-04-09:
Hi all,
I use a list to display a list of arrays string. Since the array length is longer than screen height, all the items are not shown. I expected that once I move among the item using up/down key list move up and down as user can see the focused item ( same issue as Drop down list  #130) .
Moreover, I need to know how can I get the focused btmn object in the list?

kisvegabor, 2018-04-10:
Hi,

The lv_list should "step" using LV_GROUP_KEY_UP/DOWN keys. And call the button's action function on LV_GROUP_KEY_ENTER.

Currently there is not explicit (simple) way to get the focused button on an list. Why do you want to do it? Maybe we can offer a workaround.

Masi2017, 2018-04-10:
@kisvegabor Yes it supposed to"step"  but it's not. that's why I was looking for a way to do that manually. It seems it has the same issue as Drop down list had before and then you resolved in new version 5.1.

Masi2017, 2018-04-10:
I found the reason, problem was because I haven't set the height . I set the height and now it step up and down with the cursor. Thanks

kisvegabor, 2018-04-10:
Let's continue the discussion in: #197

closed 2018-04-10 by kisvegabor