lvgl/lvgl #197: list_up and down doesn't work properly

Masi2017, 2018-04-09:
Hi All,

I want to use List obj, I wrote the following code, I am using the hardware keyboard. Once I move down I can not see the last item of the list , if I change the list position (    lv_obj_set_pos(list2, 10, 60);) I am not able to see the two last items and so on. I would appreciate if someone help me regarding this issue.

void lv_test_list_1(void)
{

    list2 = lv_list_create(lv_scr_act(), NULL);
    lv_obj_set_pos(list2, 10, 10);

    lv_list_add(list2, SYMBOL_FILE, "File", NULL);
    lv_list_add(list2, SYMBOL_DIRECTORY, "Directory", NULL);
    lv_list_add(list2, SYMBOL_UP, "Up", NULL);
    lv_list_add(list2, SYMBOL_DOWN, "Down", NULL);
    lv_list_add(list2, NULL, "No icon", NULL);
    lv_list_add(list2, SYMBOL_CLOSE, "", NULL);
    lv_list_add(list2, SYMBOL_UP, "Up", NULL);
    lv_list_add(list2, SYMBOL_DOWN, "1", NULL);
    lv_list_add(list2, SYMBOL_DOWN, "2", NULL);
    lv_list_add(list2, SYMBOL_DOWN, "3", NULL);
    lv_list_add(list2, SYMBOL_DOWN, "4", NULL);

    lv_obj_set_width(list2, LV_HOR_RES-80);

              g1 = lv_group_create();
              lv_indev_set_group(kb_indev, g1);

    /*Add list up/down buttons*/
    lv_obj_t *btn_up = lv_btn_create(lv_scr_act(), NULL);
  //  lv_obj_align(btn_up, NULL, LV_ALIGN_IN_BOTTOM_LEFT, 0, 10);
    lv_obj_set_width(btn_up, 50);
    lv_obj_set_pos(btn_up,10,LV_VER_RES-120);
    lv_btn_set_action(btn_up, LV_BTN_ACTION_CLICK, list_move);
    lv_obj_set_free_num(btn_up, 0);
    lv_obj_t *label = lv_label_create(btn_up, NULL);
    lv_label_set_text(label, SYMBOL_UP);

    lv_obj_t *btn_down = lv_btn_create(lv_scr_act(), btn_up);
    lv_obj_align(btn_down, btn_up, LV_ALIGN_IN_RIGHT_MID, 100, 0);
    lv_obj_set_free_num(btn_down, 1);
    label = lv_label_create(btn_down, NULL);
    lv_label_set_text(label, SYMBOL_DOWN);
//    lv_list_up(list3);
  //  lv_list_up(list4);
    lv_group_add_obj(g1, btn_up);
    lv_group_add_obj(g1, btn_down);
//    lv_group_add_obj(g1, list2);

}

/**********************
 *   STATIC FUNCTIONS
 **********************/

static lv_res_t list_move(lv_obj_t *btn)
{
    if(lv_obj_get_free_num(btn) == 0) {
        lv_list_up(list2);

    } else {
        lv_list_down(list2);
    }
    return LV_RES_OK;
}

kisvegabor, 2018-04-10:
Hi,

It seems you are using this concept a little bit differently as it was intended.
1. You don't have to use `lv_list_up/down` in this case.
2. You just need to focus on the list with the group and send LV_GROUP_KEY__UP/DOWN to it.

I added these 2 lines at the end if your `lv_test_list_1` function and it now it worked well:
```c
lv_group_add_obj(g1, list2);
lv_group_focus_obj(list2);
```

Masi2017, 2018-04-10:
In fact I added the list to the group , same as what you did, but I had problem in moving up and down when list item are more then screen height , then I checked the existing example from the lv_boards and try test that to make sure I am doing something wrong or the issue exists. But it can't move down along with the cursor same as the issue was in drop down list and then you resolved in ver 5.1 . Then I tried to call  list_up and down manually but even if I call these functions I not all the items are visible.

kisvegabor, 2018-04-10:
Hi,

I just added those two lines to your code and it worked. Can you provided a working and a not working code example?

It seems somehow I didn't get your email.
Please send it again directly to kisvegabor@gmail.com

kisvegabor, 2018-05-17:
It seems the problem already solved so I close this issue.

closed 2018-05-17 by kisvegabor