lvgl/lvgl #200: Improvements with handling includes

kl-cruz, 2018-04-11:
Lib version: Lvgl 5.1
usage: lvgl as standalone library, without lv_boards configs
issue: I have to place lv_conf.h file in particular level in directories to make library work

There are a lot of includes with mixed relative paths like:
https://github.com/littlevgl/lvgl/blob/master/lv_draw/lv_draw.h#L17
or:
https://github.com/littlevgl/lvgl/blob/cde8738f0e57fd8ce99cc95bcca47562dca88e84/lv_themes/lv_theme_default.h#L16

Includes should point to root point in lvgl hierarchy or point directly to files like:
#include <lv_conf.h>

Consider to use <> brackets to point compiler to look at files outside current directory.

Please let me know if You are working on it now. If not I can take over this issue and propose solution.

kisvegabor, 2018-04-12:
Hi,

It seems to me the only questionable file is `lv_conf.h` because it is out of the lvgl folder. All the internal files should work with relative includes because their location won't change. Do you agree?

I have chosen the use of relative include path for `lv_conf.h` too to avoid extra compiler settings (adding include path manually). The current method ensures at least one working project structure without additional settings.

You are right it's not flexible enough. However I think it's important to keep the current "simple usage" opportunity. So I suggest to make possible to choose between relative path (`"../../lv_conf.h"`) and added include path (`<lv_conf.h>`) using a compiler define. For example if you added a "system level" define (`-D` in gcc) like `LV_CONF_OUTSIDE` it would choose between the two includes. )It's not good name but didn't have better idea now :) )

```c
#ifdef LV_CONF_OUTSIDE
#include <lv_conf.h>
#else
#include "../../lv_conf.h"
#endif 
```

Nobody is working it right now. An update like this would be highly appreciated!

kl-cruz, 2018-04-12:
I do not want to start a revolution here ;) 
The solution proposed by You is good and should make lvgl config file flexible when working as a part of other project. I agree.
I also suggested to start another includes (lvgl ones) from root of the lvgl directory. Why?

- It could be easy to maintain when some files are moving to another multilevel directory. With path relative to current directory it is easy to lost yourself and meet lot of pain during integration.
- Easy way to include lvgl in one include entry during compilation.
- Is not as fragile as relative paths. Consider situation when someone will put all the files in `src` directory and headers to `inc`
- It is easier to find header this way :)
- just imo looks better

My suggestion here is to start every file from the root. For example change:

https://github.com/littlevgl/lvgl/blob/5b9c9c4833f71ad9a8aaf16500235a3d130b764c/lv_draw/lv_draw.h#L17

to
```c
#ifdef LV_CONF_NEW_INC_PATH
#include <lv_misc/lv_txt.h>
#else
#include "../lv_misc/lv_txt.h"
#endif 
 ```

kisvegabor, 2018-07-07:
Hi,

There were some updates on this topic:
1. In `dev-5.2` branch you can define  `LV_CONF_INCLUDE_SIMPLE` to use simple (`#include "lv_conf.h"`) include of lv_conf.h (#294)
2. I was thinking about the similar include of other lvgl files too but I concluded that the relative includes are required because of the compilers where you don't have opportunity to add include paths manually (e.g. Arduino). So we need to maintain both the simple and relative includes. I collected the pros/cons of using both include type and switch with define: 
- Cons:
  - Relative paths still need to be maintained
  - It will result long include sections (Every include will be 6 lines)
- Pros:
  - Makes possible to move all the headers to an `inc` folder. However, as the whole library can be included with `lvgl/lvgl.h` I'm not sure it has real benefits to restructure the library this way.

So for now I vote for to leave the relative path for the library headers.

As the main issue with "lv_conf.h" is solved I close this issue. If you have any ideas, remarks or suggestion feel free to reopen this issue.

Thank you for your help to improve the library!  :)

closed 2018-07-07 by kisvegabor