lvgl/lvgl #206: Redraw an an object

Masi2017, 2018-04-13:
I that any way to redraw an object after being created? Assume that I create a keyboard, then  once user type the text and insert Enter key, a message box including two option "Edit" and "Save" pops up, if Edit is pressed user is returned to the keyboard , Is that any way to redraw the keyboard?

kisvegabor, 2018-04-13:
If you mean by "Enter" the "Ok" button then yes. Clicking the "Ok" or "Hide" button will delete the keyboard by default but if you assign an `ok_action` then you can control what to happen on click. You can hide the keyboard or animate it out.

Use: `lv_kb_set_ok_action()`

Masi2017, 2018-04-13:
@kisvegabor Oh no I mean assume that I create an object which has common area with the previous object. Once I delete the second object , the common area will be cleared so I have to redraw the previous one. I was looking for a way not to delete and create the previous object.

Masi2017, 2018-04-13:
@kisvegabor I think I should focus on the previous object only!

kisvegabor, 2018-04-16:
The managing of redrawing should be done by the library automatically. Please send an example code to see what happens.

kisvegabor, 2018-05-17:
It seems the issue is solved so I close it. 
If you have any remarks in this topic feel free to reopen it!

closed 2018-05-17 by kisvegabor