lvgl/lvgl #207: Remove all added object from group

Masi2017, 2018-04-13:
Is that any way to remove all added objects from a group?

kisvegabor, 2018-04-13:
There is no API function for it. However it should work (not tested):
```c
    lv_group_t* g1;
    lv_obj_t ** obj;
    LL_READ(g1->obj_ll, obj) {
        lv_obj_del(*obj);
    }
```

Masi2017, 2018-04-13:
thanks I'll test it.

Masi2017, 2018-04-19:
Hi @kisvegabor , you mean adding this function might remove all objects added to the group? 

void lv_group_remove_all (lv_group_t* g1) {

    lv_obj_t ** obj;
    LL_READ(g1->obj_ll, obj) {
        lv_obj_del(*obj);
    }
}

I was looking for make group empty. Problem is when I run lv_obj_del to delete the object and all its children, I have to remove them from the group if they've been already added to a group since this is not  done in lv_obj_del .
Another question that I have is if I delete a group completely (lv_group_ , still I need to remove objects from that (of course before depleting the group) ?

kisvegabor, 2018-04-20:
Hi,

Oh, it seems I misunderstood you. Yes, my suggestion deleted all the object in a group, not from the group.

> I have to remove them from the group if they've been already added to a group since this is not done in lv_obj_del .

It is already solved in `beta`

> Another question that I have is if I delete a group completely (lv_group_ , still I need to remove objects from that (of course before depleting the group) ?

I checked it and `lv_group_del` doesn't remove the objects from the group but it ought to be. It's a bug now which should be fixed.

Masi2017, 2018-04-20:
@kisvegabor thanks I'll be waiting for the new release .

kisvegabor, 2018-05-16:
Hi,

I fixed it in the `beta` branch. Now `lv_group_del` should remove the objects from the group too.
Can you make a quick test?

Masi2017, 2018-05-17:
Hi,
I would test that during the weekend.
Is that possible to remove the object from group once object deleted? I mean calling lv_obj_del , remove object from the group too.

kisvegabor, 2018-05-18:
Great, thank you!

I already added a fix to the `beta` branch for `lv_obj_del` to remove the object from the group.

Masi2017, 2018-05-18:
Thanks, I'll test that and let you know the result.

Masi2017, 2018-05-25:
I haven't had the chance to test this but I'll test it as soon as I find the time .

Masi2018, 2018-06-17:
@kisvegabor Hi, sorry for the delay.
I tested but it doesn't work fine. The problem is if you create an object and delete the object and then create another object ( different object type) lv_group_focus_next doesn't work fine and system crashes (I tested the code with the previous version and it worked fine , seems that new changes causes another bug). here is my test code :

lv_obj_t *InboxScreen;
lv_group_t *g1

void test();

void test3(){

  lv_group_set_focus_cb(g1,NULL);
  lv_obj_del(InboxScreen);

  test();
}
void test2() {

    InboxScreen=lv_obj_create(NULL, NULL);
    lv_scr_load(InboxScreen);

    lv_obj_t *list=lv_list_create(InboxScreen, NULL);

    lv_list_add(list, SYMBOL_LIST ,"Predefined List", NULL);

    lv_group_add_obj(g1,list);

    vTaskDelay(pdMS_TO_TICKS(100));
    lv_group_set_focus_cb(g1,test3);

}
void test1() {

  lv_group_set_focus_cb(g1,NULL);
  lv_obj_del(InboxScreen);

  test2();
}

void test() {

  mainMenuKeypadOpsInit();

  lv_indev_set_group(kb_indev, g1);
  InboxScreen=lv_obj_create(NULL, NULL);
  lv_scr_load(InboxScreen);

  lv_obj_t *btn1= lv_btn_create(InboxScreen, NULL);
  lv_group_add_obj(g1,btn1);

  currentObj=btn1;
  lv_group_set_focus_cb(g1,test1);
}

void UITask(void *pvParameters) {

  g1 = lv_group_create();

test();

  while (1) {
    vTaskDelay(1);
    lv_task_handler();
  }

}

kisvegabor, 2018-06-17:
Hi,

I tested it and found the bug. It was only present when you deleted the last object from the group.

I pushed the fix to the `beta` branch.

Masi2018, 2018-06-17:
@kisvegabor Thank you I tested , it was fine.

kisvegabor, 2018-06-17:
Great! 
Thank you for the testing!

closed 2018-06-17 by kisvegabor

ali-rostami, 2019-02-03:
Hi
I want to do the same thing.
 When I navigate in the menu I want to delete the objects of the group and add the new objects -the opened menu items- I think it's not good to delete the group object and recreate it again, isn't it?
and if yes, what's the best way to do this? Isn't it better to define a function for it in lv_group.c?

kisvegabor, 2019-02-04:
Both the old and the new menu exist at the same time? If you delete the old menu, the objects will be removed from the group automatically.

ali-rostami, 2019-02-05:
There are situations that the existing member of the group won't be deleted. and I just need to assign the new object to the group.

kisvegabor, 2019-02-06:
I don't fully understand. Please open a new issue for it and describe your exact case. Some screenshots can help :)