lvgl/lvgl #208: Label alignement after text changed

sdalu, 2018-04-14:
Trying to change text after object alignement, the alignement is not recalculated 

Using the following code: "15" is correctly aligned bottom right
~~~c
    lv_obj_t *label_timer = lv_label_create(root, NULL);
    lv_label_set_text(label_timer,  "15");
    lv_obj_align(label_timer, root, LV_ALIGN_IN_BOTTOM_RIGHT, 0, 0);
~~~

Next, using this code, "1" is not fully aligned to the right
~~~c
    lv_label_set_text(label_timer,  "1");
~~~

Did I missed a function call to force alignement recalculation?

kisvegabor, 2018-04-15:
Hi,

yes, `lv_obj_align` aligns the object only once and doesn't move it again when the coordinates are changed.

So you have to call `lv_obj_align`again after `lv_label_set_text`.

sdalu, 2018-04-16:
Don't know how difficult it is to implement, but  it good be an improvement for a next major release, to be able to call a function '`lv_obj_relayout`' and to have all the recalculation done automatically. That would make code updating part of the screen easier o write.

kisvegabor, 2018-04-17:
Thank you for the good idea. I agree, it could help in a lot of cases.
 
As this feature needs some extra memory in every object (align type, x and y offset, base object) it could be an option in `lv_conf.h` to enable. E.g. `LV_OBJ_REALIGN   1`. It can enable to store the settings and add the `lv_obj_realign` function. 
It's also easy to implement.

What do you think?

sdalu, 2018-04-17:
That sound great

kisvegabor, 2018-08-04:
In v6.0 the style system will be updated which also effects positioning of objects (See #302). So it's more reasonable to implement it in v6.0.

kisvegabor, 2018-11-29:
This feature is implemented in `dev-5.3`.
See: https://github.com/littlevgl/lvgl/pull/576

closed 2018-11-29 by kisvegabor