lvgl/lvgl #210: label anim speed 1, jumping whole screen length

sdalu, 2018-04-14:
Using `lv_label_set_anim_speed(label_alert, 1)`, I've got a strange behaviour: the screen jump of the whole screen length about every 15s. For value greater than 1 behaviour follows documentation. 
It could looks like a special behaviour for the value 1, but is not documented on the web site api.

Also when scrolling long text, a pause seems present at both end, but I haven't found a function to set this wait time.

kisvegabor, 2018-04-15:
Hi,

Can you send a code example to reproduce the issue?

You are correct, now there is pause break after both sides.   We had a discussion about it here: #188
In the next release there will be a function to adjust it.

sdalu, 2018-04-16:
That's the code I use
~~~c
    lv_obj_t *root = lv_scr_act();
    lv_obj_clean(root);

    lv_obj_t *label_alert = lv_label_create(root, NULL);
    lv_label_set_style(label_alert, &style_unscii_16);
    lv_label_set_long_mode(label_alert, LV_LABEL_LONG_SCROLL);
    lv_label_set_anim_speed(label_alert, 16);
    lv_label_set_text(label_alert,  "guierguier erer gr oeihf idgui regui ergh erugheruihgurhgue g heru ghuerh guer kh!");
    lv_obj_align(label_alert, root, LV_ALIGN_CENTER, 0, 0);
~~~

kisvegabor, 2018-04-17:
Thank you!

I fixed it in `beta` branch. There was an overflow in an animation related function. Now the the max animation time is limited to 65,535 ms =~ 65 sec.

closed 2018-06-22 by kisvegabor