lvgl/lvgl #214: Simple label based dynamic ui

chruetli, 2018-04-17:
Hi
Thanks for this simple but powerful lib!
As I have a quite small display I try to use only labels and an image to display my data. I have:
-------------------
Image
Label1: Title
Label2: Value
Label3: Time
-------------------
The setup with static data works great. But when values are updated the whole layout gets messed... I cannot simply rewrite the labels.
Does the whole layout need to be redrawn? Or is there a simpler solution?

Thanks Mario

kisvegabor, 2018-04-18:
Hi Mario,

Please post the relevant parts of your code to see what's going on!

kisvegabor, 2018-04-20:
Hi,

I tested you code and worked for me fine.

When you enable ROLL mode the label jumps to tops immediately?  If yes, pleas try the ROLL function on GUI with only a single label to minimize the possible issues.

chruetli, 2018-04-20:
I tried it and I'm sorry, you are right, my example doesn't show the effect. I had to make the label "rolling" to see the problem. Please use a longer text, e.g.
```
lv_label_set_text(label2, "Measured = 1234.5678 mm/s");
```

kisvegabor, 2018-04-20:
It's still working for me. 

Can you send a small, simple code with only a label in question?

chruetli, 2018-04-20:
Hu, that's strange. I'm using version 5.1.0. Bellow a 'minimal' version...

```C
#include "lvgl/lvgl.h"
#include "lv_drivers/display/fbdev.h"

lv_task_t * refr_task;
lv_style_t styleLabel2;
lv_obj_t * label2;

lv_init();
fbdev_init();
lv_disp_drv_t disp_drv;
lv_disp_drv_init(&disp_drv);
disp_drv.disp_flush = fbdev_flush;     
lv_disp_drv_register(&disp_drv);

refr_task = lv_task_create(sysmon_task, 500, LV_TASK_PRIO_LOW, NULL);

/* Label-2 */
lv_style_copy(&styleLabel2, &lv_style_plain);
styleLabel2.text.font = &lv_font_dejavu_20;
styleLabel2.text.letter_space = 2;
styleLabel2.text.line_space = 1;
styleLabel2.text.color = LV_COLOR_MAKE(0x00,0xad,0xa5);

label2 = lv_label_create(lv_scr_act(), NULL);
lv_obj_set_style(label2, &styleLabel2);
lv_label_set_long_mode(label2, LV_LABEL_LONG_ROLL);
lv_label_set_recolor(label2, true);
lv_label_set_text(label2, "Measured = 1234.5678 mm/s");
lv_obj_set_width(label2, 160);
lv_obj_align(label2, NULL, LV_ALIGN_IN_TOP_LEFT, 0, 70);

while(1) {
	lv_tick_inc(5);
	lv_task_handler();
	usleep(5000);
}

void sysmon_task(void * param)
{
}

```

kisvegabor, 2018-04-20:
For me it's still not on the top.

Please attach your lv_conf.h and show an example `sysmon_task` task as well.
Somewhere must be a difference :)

chruetli, 2018-04-20:
Sorry for the inconvenience...
The `lv_conf.h` is below, the `sysmon_task` is empty and only a placeholder at the moment.
Could the framebuffer driver be the problem?

```C
/**
 * @file lv_conf.h
 *
 */

// #if 0 /*Remove this to enable the content (Delete the last #endif too!)*/

#ifndef LV_CONF_H
#define LV_CONF_H

/*----------------
 * Dynamic memory
 *----------------*/

/* Memory size which will be used by the library
 * to store the graphical objects and other data */
#define LV_MEM_CUSTOM      0                /*1: use custom malloc/free, 0: use the built-in lv_mem_alloc/lv_mem_free*/
#if LV_MEM_CUSTOM == 0
#define LV_MEM_SIZE    (32U * 1024U)        /*Size memory used by `lv_mem_alloc` in bytes (>= 2kB)*/
#define LV_MEM_ATTR                         /*Complier prefix for big array declaration*/
#define LV_MEM_AUTO_DEFRAG  1               /*Automatically defrag on free*/
#else       /*LV_MEM_CUSTOM*/
#define LV_MEM_CUSTOM_INCLUDE <stdlib.h>   /*Header for the dynamic memory function*/
#define LV_MEM_CUSTOM_ALLOC   malloc       /*Wrapper to malloc*/
#define LV_MEM_CUSTOM_FREE    free         /*Wrapper to free*/
#endif     /*LV_MEM_CUSTOM*/

/*===================
   Graphical settings
 *===================*/

/* Horizontal and vertical resolution of the library.*/
#define LV_HOR_RES          (160)
#define LV_VER_RES          (128)
#define LV_DPI              100

/* Size of VDB (Virtual Display Buffer: the internal graphics buffer).
 * Required for buffered drawing, opacity and anti-aliasing
 * VDB makes the double buffering, you don't need to deal with it!
 * Typical size: ~1/10 screen */
#define LV_VDB_SIZE         (20 * LV_HOR_RES)  /*Size of VDB in pixel count (1/10 screen size is good for first)*/
#define LV_VDB_ADR          0                  /*Place VDB to a specific address (e.g. in external RAM) (0: allocate automatically into RAM)*/

/* Use two Virtual Display buffers (VDB) parallelize rendering and flushing (optional)
 * The flushing should use DMA to write the frame buffer in the background*/
#define LV_VDB_DOUBLE       0       /*1: Enable the use of 2 VDBs*/
#define LV_VDB2_ADR         0       /*Place VDB2 to a specific address (e.g. in external RAM) (0: allocate automatically into RAM)*/

/* Enable anti-aliasing (lines, and radiuses will be smoothed) */
#define LV_ANTIALIAS        1       /*1: Enable anti-aliasing*/

/*Screen refresh settings*/
#define LV_REFR_PERIOD      50    /*Screen refresh period in milliseconds*/
#define LV_INV_FIFO_SIZE    32    /*The average count of objects on a screen */

/*=================
   Misc. setting
 *=================*/

/*Input device settings*/
#define LV_INDEV_READ_PERIOD            50                     /*Input device read period in milliseconds*/
#define LV_INDEV_POINT_MARKER           0                      /*Mark the pressed points  (required: USE_LV_REAL_DRAW = 1)*/
#define LV_INDEV_DRAG_LIMIT             10                     /*Drag threshold in pixels */
#define LV_INDEV_DRAG_THROW             20                     /*Drag throw slow-down in [%]. Greater value means faster slow-down */
#define LV_INDEV_LONG_PRESS_TIME        400                    /*Long press time in milliseconds*/
[... 164 lines cut ...]
#define LV_DDLIST_ANIM_TIME     200     /*Open and close default animation time [ms] (0: no animation)*/
#endif

/*Roller (dependencies: lv_ddlist)*/
#define USE_LV_ROLLER    1
#if USE_LV_ROLLER != 0
#define LV_ROLLER_ANIM_TIME     200     /*Focus animation time [ms] (0: no animation)*/
#endif

/*Slider (dependencies: lv_bar)*/
#define USE_LV_SLIDER    1

/*Switch (dependencies: lv_slider)*/
#define USE_LV_SW       1

#endif /*LV_CONF_H*/

// #endif /*Remove this to enable the content*/
```

```C
void sysmon_task(void * param)
{
}
```

chruetli, 2018-04-20:
Ok, thank you, will try this. Unfortunately I have to wait until Tueseday. 
Enjoy the weekend!

kisvegabor, 2018-04-20:
Ok. I hope something will turn out!
I will be on a holiday on next week. So won't be able to be active here. 
I will be back on April 30.

chruetli, 2018-04-24:
Hope you had nice holidays!! Found the problem thank to your suggestion about printing the coordinates in the `disp_flush` function. My framebuffer device driver didn't handle the coords right. Now it works as it was expected!

Thanks!

closed 2018-04-24 by chruetli

kisvegabor, 2018-04-30:
Great! :)