lvgl/lvgl #215: littlev task locks?

ramonschepers, 2018-04-17:
Just wondering, does the lvgl task have locks (semaphores/mutexes)?

Looking at the code, I assume not. 

Would be nice to have for io port busy flags and wait for other tasks to complete

kisvegabor, 2018-04-18:
Hi,

If you mean by semaphore:
1. Lock a task if the semaphore is taken
2. Stop the task running
3. Continue running from the "lock" when the semaphore become free

then it could be already a full featured OS. The `lv_task` system is very light weighted and can only call functions periodically but call the higher priority functions earlier.
If you need so sophisticated management I can suggest use e.g. FreeRTOS. (LittlevGL can be integrated easily)

But if you mean: Check the semaphore and just return if it is taken, then it is possible to provide an API for it.

kisvegabor, 2018-07-14:
As there wasn't activity here for months I close this issue.
If you have a similar question feel free to comment here and reopen the issue!

closed 2018-07-14 by kisvegabor