lvgl/lvgl #216: lv_ddlist_close wrong decleration

Masi2017, 2018-04-18:
Hi all,
In lv_ddlist.h there is "lv_ddlist_close_en" whereas in lv_ddlist.c it's defined as "lv_ddlist_close", I think it needs to be corrected.
Also I had to make some changes otherwise if user delete ddlist in user action system crashes 
in "lv_ddlist_release_action" I added the following code "if (ddlist==NULL) return LV_RES_OK "
In lv_ddlist_signal I changed a part of code as following :

else if(c == LV_GROUP_KEY_ENTER  || c == LV_GROUP_KEY_ENTER_LONG) {
            if(ext->opened) {
                ext->sel_opt_id_ori = ext->sel_opt_id;
                ext->opened = 0;
                if(ext->action) ext->action(ddlist);
else   lv_ddlist_refr_size(ddlist, true);
            }
            else {
                ext->opened = 1;
lv_ddlist_refr_size(ddlist, true);
            }

 Please let me know if this cause any problem (because of the above changes if user defined the ddlist action, lv_ddlist_close should be called in user action if it's needed since lv_ddlist_refr_size will not called automatically )

kisvegabor, 2018-04-18:
Hi,

thank you for the remark! Can you create a Pull request from your modification to better see what changed?

Masi2017, 2018-04-19:
@kisvegabor Ok sure, I haven't done the pull request but let's see how should I do that.

Masi2017, 2018-04-20:
@kisvegabor I uplaoded my modified lvgl lib here:
https://github.com/Masi2017/lvgl/tree/master
But I couldn't find how to do a pull request! I may spend sometime to learn about it on weekend!