lvgl/lvgl #218: Change Mbox btn size

Masi2017, 2018-04-19:
Hi ,
I was wondering how I can change the mbox btn size? using control character I am able to adjust btn width related to the other btns but I need to set my desired width.

Masi2017, 2018-04-20:
@kisvegabor Thank you but I need to change the button width and also it comes from text width. Assume that I want to display "Message sent successfully" and add one one confirmation "OK" button , the width of confirmation button would be quite big and I like to make it smaller. Also same in Button Matrix , I don't know how change the Button width individually ( not related to the other buttons in the row)

kisvegabor, 2018-04-20:
You can do it with a trick by adding hidden buttons. (\21x control byte)

Masi2017, 2018-04-20:
@kisvegabor Thank you good idea!

Masi2017, 2018-05-09:
@kisvegabor The problem using left and right key , hidden key also is selected and seems that cursor goes beyond the keys.

kisvegabor, 2018-05-09:
Yes, really it also "selects" the hidden buttons which seems to be a bug or at least not intended.

Button matrix supports only relative width adjusting. 

Can you send a simple sketch about the desired result? Likely it can be reached with the Button matrix.

Masi2017, 2018-05-18:
I want to have a message box with a confirmation btn. If I use mbox and add a btn to mbox , btn width would be as big as mbox width whereas I am looking for something similar to the following picture which I have used button  object so I was able to change width and height independently. 
[image]

kisvegabor, 2018-05-18:
I see. To reach this you can modify the `LV_MBOX_STYLE_BTN_BG`

```c
	static lv_style_t mbox_btns_bg;
	lv_style_copy(&mbox_btns_bg, &lv_style_plain_color);
	mbox_btns_bg.body.padding.hor = 50;

	const char * btns_str[] = {"Ok", ""};
	lv_obj_t * mb = lv_mbox_create(lv_scr_act(), NULL);
	lv_mbox_add_btns(mb, btns_str, NULL);
	lv_mbox_set_style(mb, LV_MBOX_STYLE_BTN_BG, &mbox_btns_bg);
```

I intentionally used a non-transparent style to see what happens. 
[image]

Masi2017, 2018-05-18:
Oh great, thanks!

Masi2017, 2018-06-07:
I have some problem to use control character. For example if I want to have 3 buttons as "Send", "Attach", "Cancel" in the mbox, what would be the best control character for each?

kisvegabor, 2018-06-07:
I checked the documentation and it was mistyped, sorry. I corrected it. https://littlevgl.com/object-types/button-matrix-lv_btnm

So if the control character is in octal format (`\xxx` in C) then it must start with `\2xx`.
By default all buttons have the width of "1 unit". But with the last number in the control byte you can modify it. E.g.  3  unit: `\2x3`. It will make the button 3 time wider the the others. It always proportional so doesn't matter how much button you have in row the relative width will remain. 
  
Here is an example:
```c
    static const char * btns[] = {"\203Triple","\202Double","Simple",""};
    lv_obj_t * mb = lv_mbox_create(lv_scr_act(), NULL);
    lv_mbox_add_btns(mb, btns, NULL);
    lv_obj_set_width(mb, 300);
    lv_obj_align(mb, NULL, LV_ALIGN_CENTER, 0, 0);
```
[image]

Masi2017, 2018-06-08:
Hi, thank you very much. This issue can be closed too.

kisvegabor, 2018-06-09:
Your welcome!

closed 2018-06-09 by kisvegabor