lvgl/lvgl #221: [Future Request] Possible to get a C++ version of this librairy?

caveman1996, 2018-04-20:
Hi, we love the librairy and up to now, it will be very usefull. Thought, with the C++ code being more utilised, we have a lot of C++ librairies and it would be nice to have a C++ version of this librairy. We know it can need a lot of time to convert it, but is it an option for the future? Thanks

kisvegabor, 2018-04-20:
Hi,

The library is already compatible with C++. At least according my experiences so far. For example I compiled it for Arduino which uses C++.

caveman1996, 2018-04-20:
Thanks for the fast answer, as always :grinning: 

Well, when we compile our code, there is more than 500 errors comming from the librairy. The erros message was about conversion from void* to (something) *char if I remember correctly. Maybe we didn't do our job correctly, but it look liked the code wasn't well takien by the ARM c++ compiler.

hreichert, 2018-04-20:
I can confirm that the lib works with C++
(MBed OS 5 with ARM GCC)

caveman1996, 2018-04-20:
Ok at least it's nice to ear. How do you compile the lib with C++ code? Is there anything you need to change?

Vonger, 2018-04-27:
C is 100% compatible with C++ 
Just ignore the warning 😄

caveman1996, 2018-04-27:
C isn't 100% compatible from my understanding. It seems that you need to convert some parameters. If it was just warning I wouldn't get help here, we have more than 500 **errors** coming from the librairy. Thanks

kisvegabor, 2018-04-30:
As I know the most important thing to make C compatible with C++ it the use of
```c
#ifdef __cplusplus
extern "C" {
#endif
```

macros in the `.h` files. It is done in every file of LittlevGL and it worked for me in all C++ environments I tried so far.

kisvegabor, 2018-07-14:
As there wasn't activity in this topic for months I close this issue. 
If you face problems when using LittlevGL in C++ environment feel free to comment here.

closed 2018-07-14 by kisvegabor