lvgl/lvgl #223: keep List focused button

Masi2017, 2018-04-20:
Hi again!
Is there anyway to keep the selected record (button) from list after return back to the list in a group of object. I mean assume that I added four item in a list and then user select the second one and a message box appears so now mbox has been focused , if I press the prev key , list is focused again but the selected record from the list is reset and first records be selected not the second one.

kisvegabor, 2018-04-20:
Hi,

It's really a "lifelike" situation but unfortunately I can't say a built-in method for it. 

As workaround I can tell that  the `lv_list` knows which button is "active" from the button's state. The active button has `LV_BTN_STATE_TGL_REL` while the others have `LV_BTN_STATE_REL`.
However I didn't test it but if you have only 4 buttons you can set their states manually to trick the list.
Only set the desired button's state to `LV_BTN_STATE_TGL_REL` and the rest to `LV_BTN_STATE_REL`

By the way I will be on a holiday on next week. So I won't be able to be active here at this time but I hope everything will be fine here. :) I will be back on April 30.

kisvegabor, 2018-05-18:
Yes!

kisvegabor, 2018-07-30:
Hi,

I added `lv_list_set_btn_selected(list, btn)` and `lv_list_get_btn_selected(list)` function.

I tested it but if it doesn't work, please reopen this issue.

Thank you for the suggestion! :)

closed 2018-07-30 by kisvegabor