lvgl/lvgl #226: how to hide the buttons on tabview?

liubingyan123, 2018-04-27:
as the titile says, I created a tabview consist of three tab, shown as following pictures, I want to hide the tab buttons or make them invisiable, and also the slider on yhe right side, but failed, please offer me some advices,thank you!
 
[image]
[image]

kisvegabor, 2018-04-30:
Hi,

Now you can't hide the "header". It's height comes from the **font height + body.padding.ver attributes of LV_TABVIEW_STYE_BTN_BG and LV_TABVIEW_STYE_BTN_REL **
As a workaround I can suggest to set a bigger height for the tabview and set negative y position to adjust the header out of the screen or parent object.
  
To hide the scrollbar you can apply this for every page you add to the tabview:
 ```c
    lv_obj_t * p1 = lv_tabview_add_tab(tv, "first");
    lv_page_set_sb_mode(p1, LV_SB_MODE_OFF);
```

liubingyan123, 2018-05-08:
ok, I'll try it, thank you very much!

closed 2018-05-08 by liubingyan123

zhangshengping, 2019-03-08:
@kisvegabor  Are you planning to add this feature?

kisvegabor, 2019-03-08:
See https://github.com/littlevgl/lvgl/issues/628#issuecomment-470959880