lvgl/lvgl #227: Drop up List?

gjohnson0720, 2018-04-27:
I'd like to use drop-down list functionality only have the "button" along the bottom of my display and drop-up the list when it's pressed.

kisvegabor, 2018-05-02:
Hi,

Right now the `lv_ddlist` object doesn't support "drop-up" functionality. 

I suggest to create a normal button (lv_btn) at the bottom of your screen and a normal list (lv_list) and in the click action of the button (`lv_btn_set_action(my_btn_1, LV_BTN_ACTION_CLICK, my_click_action);`) hide/unhide the list (`lv_obj_set_hidden(my_list_1, true/false)`)

kisvegabor, 2018-09-18:
As there was no activity here for a while I close it to keep the issues clear.

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2018-09-18 by kisvegabor