lvgl/lvgl #228: Draw circle

RocFan, 2018-04-29:
How to draw a circle at position of an object?

kisvegabor, 2018-05-02:
Hi,

To draw a circle set `LV_RADIUS_CIRCLE` as radius in the object's style:
```c
my_style.body.radius = LV_RADIUS_CIRCLE;
```

RocFan, 2018-05-02:
Thanks，Actually I want to know how to draw an arc, like that.
[image]

RocFan, 2018-05-02:
I can use the "border" of style, right? but how to draw 1%?

kisvegabor, 2018-05-02:
Unfortunately LittlevGL doesn't support arc drawing at this moment. (There are plans to add this feature)
And now the border is mono-color.

To crate sophisticated arcs and don't need high resolution (e.g. 10 states is okay) then I suggest to build it from small images and hide which is not required.

RocFan, 2018-05-03:
Thanks, I will try.

kisvegabor, 2018-05-17:
Arc drawing become a planned feature in an upcoming release.

We can continue the discussion about it here: #255

closed 2018-05-17 by kisvegabor