lvgl/lvgl #229: Font Converter feature request

DeonMarais64, 2018-04-30:
Hi Gábor,
Would it be possible to provide an optiion on your font converter to make all glyphs in a conversion mono spaced. IE All glyphs be the same pixel width.

I thought of two alternatives to the above in order not to waste code space,
1. Either have another member in the lv_style_t->text struct to make each emmision mono spaced accupying a particular number of pixels or to somehow encode this intention to the lv_style_t->text->letter_space field.
2. Add a member to the _lv_font_struct for the same reason. The advantage of method 1 above would be that this becomes style flexible.

If I am missing something (I am still learning the ropes with lvgl) and this can be acheived via other means then I would appreciate any guidance you are willing to offer.

Best Regards,
Deon Marais

DeonMarais64, 2018-05-01:
Thanks for that

kisvegabor, 2018-06-22:
Hi,

I added the **monospace** feature to the font converter. As you suggested I added a property to `lv_font_t` and rendered the letters according to this property.

The updates are in the `dev-5.2` branch.

closed 2018-06-22 by kisvegabor