lvgl/lvgl #230: TextArea cursor blinking invalidates TA even when invisible

fpw, 2018-04-30:
Hello,

I'm evaluating LittlevGL for a project and noticed something that hogs the CPU more than I thought: The blinking cursor of a textarea invalidates the whole textarea, causing blit calls in the driver with big regions. This is fine in my opinion if the textarea is currently visible.

However, I'm using a textarea inside a tabview and if I select a tab without a textarea, the components are still redrawn because the blinking cursor of the textarea in another tab invalidates its region. This is rather unexpected. As a workaround, I disable the cursor when switching tabs.

Regards

Folke

kisvegabor, 2018-05-02:
Hi,

As you said now the whole text area is blinking because of the cursor. (which should be improved in the future ) However it shouldn't blink if it's not visible.

I tested it using this simple code:
```c
    lv_obj_t * tv = lv_tabview_create(lv_scr_act(), NULL);
    lv_obj_t * p1 = lv_tabview_add_tab(tv, "first");
    lv_obj_t * p2 = lv_tabview_add_tab(tv, "second");
    lv_obj_t * ta = lv_ta_create(p2, NULL);
```
And I enabled `LV_INDEV_POINT_MARKER` in `lv_conf.h`. I pressed (not dragged) the first page where the text area should be if the second tab was loaded. 
I saw that the red points are remained there which means the area is not redrawn.

If I did the same on the Text area (pressing it on the second tab) and the point disappeared on every cursor blink.

From this it seems working for me. Or did I misunderstood something?

fpw, 2018-05-02:
I tested it with debug outputs in the blit call of the display driver, it is asked to blit the whole area of the invisible text area even though nothing of the content changed, maybe that's why the red dots are not visible?

kisvegabor, 2018-05-02:
It's strange because LittlevGL should ignore the invalidation of the invisible areas. In other words even if the TextArea invalidates itself (while an other tab is active) LittlevGL will discard this invalidation.

So will the red dots remain there if you touch the place of the text area while it is on an other tab?

fpw, 2018-05-03:
I was able to find the real problem using the red dots. My application is a tablet computer that has multiple apps like on a smartphone. I create invisible GUI containers for each app so that I open and close apps by hiding or showing their container. Deleting and re-creating the content is not an option because the containers should keep their content.

Now if one of the apps uses a textarea, I get re-blits even if the app is not currently active. So the problem was not related to the tabs, but to invisible containers. Here's example code that really shows the problem:

`        // Container for first application
        lv_obj_t *c1 = lv_cont_create(lv_scr_act(), NULL);
        lv_obj_t *l1 = lv_label_create(c1, NULL);
        lv_obj_set_click(l1, 1);
        lv_label_set_text(l1, "Click me");

        // Container for second application
        lv_obj_t *c2 = lv_cont_create(lv_scr_act(), NULL);
        lv_obj_set_hidden(c2, 1); // application not currently active
        lv_ta_create(c2, NULL);
`

The second container is hidden but it still causes invalidation of the textarea's area in the currently visible container.