lvgl/lvgl #231: Is there any manual books of the GUI?

turoksama, 2018-05-01:
Hi, is there any **# manual** books of the GUI?

closed 2018-05-02 by turoksama

turoksama, 2018-05-11:
Hi,
    Well, a "book" means a hands-on real paper document. I think it is a more effective way to learn(notes, compares at the same time, real-view, mostly do not need a 32 inch screen that makes you fuzzy).
Thanks for the tool you've mentioned. I manually collected the documents and put it together into a pdf, I would share if you like.

kisvegabor, 2018-05-11:
It would be great if you shared your pdf! 
I think a lot of people would be grateful for It! :)

turoksama, 2018-05-14:
Well, I am sending you a copy, please check for it in your email box.

kisvegabor, 2018-05-16:
I got it thank you very much.
I saw that you collected the documentation of the object types. If it helps I can do it with a HTML->PDF converter using a PHP script. Maybe it's better to generate with a script and also easier to update :)