lvgl/lvgl #235: Act very slow in an RTOS.

turoksama, 2018-05-06:
Hi, I recently moved the lv_task_handler() to a task in FreeRTOS(v9.0, TICK_RATE_HZ is 100Hz, 2 tasks with an empty task) looks like this:
/* task_gui_lvgl function */
void task_gui_lvgl(void const * argument)
{
	/* USER CODE BEGIN task_gui_lvgl */
	lv_init();
	lv_drv_disp_init();    //display driver init
	lv_drv_indev_init();  //touch input driver init
	demo_create();
	/* Infinite loop */
	for (;;)
	{
		lv_task_handler();
	}
	/* USER CODE END task_gui_lvgl */
}
/* task_empty function */
void task_empty(void const * argument)
{
	/* Infinite loop */
	for (;;)
	{
	}
	/* USER CODE END task_empty */
}

LVGL config:
/* Horizontal and vertical resolution of the library.*/
#define LV_HOR_RES          (800)
#define LV_VER_RES          (600)
#define LV_DPI              72

/* Size of VDB (Virtual Display Buffer: the internal graphics buffer).
 * Required for buffered drawing, opacity and anti-aliasing
 * VDB makes the double buffering, you don't need to deal with it!
 * Typical size: ~1/10 screen */
#define LV_VDB_SIZE         (LV_HOR_RES * LV_HOR_RES)  /*Size of VDB in pixel count (1/10 screen size is good for first)*/
#define LV_VDB_ADR          0xC0800000                 /*Place VDB to a specific address (e.g. in external RAM) (0: allocate automatically into RAM)*/

/* Use two Virtual Display buffers (VDB) parallelize rendering and flushing (optional)
 * The flushing should use DMA to write the frame buffer in the background*/
#define LV_VDB_DOUBLE       0       /*1: Enable the use of 2 VDBs*/
#define LV_VDB2_ADR         0       /*Place VDB2 to a specific address (e.g. in external RAM) (0: allocate automatically into RAM)*/

/* Enable anti-aliasing (lines, and radiuses will be smoothed) */
#define LV_ANTIALIAS        1       /*1: Enable anti-aliasing*/

/*Screen refresh settings*/
#define LV_REFR_PERIOD      50    /*Screen refresh period in milliseconds*/
#define LV_INV_FIFO_SIZE    32    /*The average count of objects on a screen */

framebuffers & vdb in external sdram(32bit), STM32f429.
It acts at least 10 times slower now than the naked one before(benchmark: around 10 ms without any effect :(  ).

Did I miss anything?

closed 2018-05-06 by turoksama

kisvegabor, 2018-05-07:
Hi,

Why did you closed the issue? Is it solved?

Anyway, I noticed two things:

1. You never yield in `task_gui_lvgl` task.

```c
	for (;;) {
		lv_task_handler();
        	 vTaskDelay( 5 / portTICK_PERIOD_MS );
	}
```

2. I can't see where do you all `lv_tick_inc()`

turoksama, 2018-05-11:
Hi,
    Yes，It is solved.
    lv_tick_inc(1);  is implemented in systick handler which is the same with FreeRTOS， and because of 100Hz of its heart beats，the time slice would be 10ms that the problem was！
    Now， I use a timer to generate a 1ms. Well,it maybe still use that with lv_tick_inc(10); .
It should be noticed that the time resolution is 10x than before, may  it affected the animation function?
I am reading the source to find out how the sysytem works.
Sincerely,
Thanks a lot!

kisvegabor, 2018-05-11:
Hi,

Thank you for the feedback!
If you apply `lv_tick_inc(10);` then it won't affect the timing of the library.

turoksama, 2018-05-14:
Hi,
That's wonderful.Thanks a lot.